ProudElm/solaredgeoptimizers #1: support for multiple inverters?

victorclaessen, 2022-08-07:
Hi,

Do you think it would be possible to extract information for the optimizers from multiple inverters in the same account?

Best,

Victor

ProudElm, 2022-08-08:
Hi,

I would think so yes, but i dont have any examples on how this would be structured within the JSON response. I would need an example to verify this.

I'm on hollidy now, when i'm home i will look into this issue.

ProudElm, 2022-08-30:
@victorclaessen do you have multiple inverters? If so, could you DM me your JSON response from this URL:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<YOURSITEID>/layout/logical

You need to login with your monitoring credentials.

victorclaessen, 2022-08-30:
I have two inverters now, and in a few weeks hope to have a third.

I've managed to find the json. (For other readers: I had to insert my site id in between the double slash [between api/sites/ and /layout/logical] in the URL.)

I don't know how to DM on github so I've sent it to the email address you've used to check in the files in this repository. I hope that's allright.

ProudElm, 2022-08-30:
Oh sorry, i failed to add that to the URL, but you've got it.

Yep, that was good. I got the json and i will take a look! Thanx!

ProudElm, 2022-09-02:
I've updated the code, it should work now. Please let me know if it does.

I've tested the JSON you've send me. The code should work, but i could not test it in Home Assistant.

victorclaessen, 2022-09-02:
Ok, it seems to find the two inverters. And after about 20 minutes, I got the first data! Power for all eight panels (two inverters, four panels per inverter). 👍 :-)

Possible improvements:
* It looks like it's not using the full panel id (i.e. 1.1.1-1, 1.1.1-2, 1.1.1-3, 1.1.1-4) but just **1.1.1** four times (and then four times 2.2.1). At least that's what it looks like in the log file (see below).
* There are now eight sensors with the name "Power". Would it be easy to (automatically) rename them "Power 1.1.1-1" and "Power 2.2.1-4", etc?

```
2022-09-02 17:25:46.638 INFO (MainThread) [homeassistant.setup] Setting up solaredgeoptimizers
2022-09-02 17:25:46.639 INFO (MainThread) [homeassistant.setup] Setup of domain solaredgeoptimizers took 0.0 seconds
2022-09-02 17:25:47.555 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solaredgeoptimizers
2022-09-02 17:25:53.845 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers found to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 1.1.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 1.1.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 1.1.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 1.1.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 2.2.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 2.2.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 2.2.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: Panel 2.2.1 to Home Assistant
2022-09-02 17:25:53.846 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Done adding all optimizers. Now adding sensors, this may take some time!
2022-09-02 17:25:53.849 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current
2022-09-02 17:25:53.849 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage
2022-09-02 17:25:53.849 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power
2022-09-02 17:25:53.850 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage
2022-09-02 17:25:53.850 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current_2
2022-09-02 17:25:53.850 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage_2
2022-09-02 17:25:53.851 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power_2
2022-09-02 17:25:53.851 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage_2
2022-09-02 17:25:53.851 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current_3
2022-09-02 17:25:53.851 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage_3
2022-09-02 17:25:53.852 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power_3
2022-09-02 17:25:53.852 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage_3
2022-09-02 17:25:53.852 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current_4
2022-09-02 17:25:53.853 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage_4
2022-09-02 17:25:53.853 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power_4
2022-09-02 17:25:53.853 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage_4
2022-09-02 17:25:53.854 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current_5
2022-09-02 17:25:53.854 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage_5
2022-09-02 17:25:53.855 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power_5
2022-09-02 17:25:53.855 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage_5
[... cut ...]
2022-09-02 17:25:53.858 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current_8
2022-09-02 17:25:53.858 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage_8
2022-09-02 17:25:53.858 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power_8
2022-09-02 17:25:53.859 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage_8
```

victorclaessen, 2022-09-09:
Thanks for that last update 98bdfc31ac7bbe0ed108f3b2adf871ea21b3aa91. Now everything shows up with the proper name and is identifiable. :-)

closed 2022-09-09 by victorclaessen