ProudElm/solaredgeoptimizers #2: Integration does not shown any devices or entities

frank8the9tank, 2022-08-12:
When adding the integration there are no entities.

steps i took:
added custom url/integration trough hacs.
added integration to home assistant
integration shows up in the integrations screen.
But no entities or devices are made under the integration.
also not after a couple of hours, nothing is shown.
There is nothing is the logs.

ProudElm, 2022-08-13:
Oh that is no good. Is wil take a look at this when i'm back from out holliday.

ProudElm, 2022-08-30:
i've updated the intergration. With hopefully some more logging. Please let me know if you are able to use it or get any logging..

frank8the9tank, 2022-08-30:
I have updated the repository through hacs, (integration was removed before hand)
Restarted
Added integration again in normal settings screen in home assistant.
But again no entities or devices,
see here:
[image]
I have this in the logs:
```
2022-08-30 17:27:47.582 INFO (MainThread) [homeassistant.setup] Setting up solaredgeoptimizers
2022-08-30 17:27:47.583 INFO (MainThread) [homeassistant.setup] Setup of domain solaredgeoptimizers took 0.0 seconds
2022-08-30 17:27:48.458 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solaredgeoptimizers
2022-08-30 17:27:49.477 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers found to Home Assistant
2022-08-30 17:27:49.477 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Done adding all optimizers. Now adding sensors, this may take some time!
```
I see nothing more related to this in the logs,

Questions:
1, is there anything that i have to look for specifically in the log?
2, i have a solaredge account with multiple sites, could that be a cause of not adding the optimizers?
3, what can i do to help futher?

ProudElm, 2022-08-30:
Ah ok.. I think this is related with issue #1. When more then 1 inverter is present, the current code cant handle that :-( Im working on a solution.

But im not sure if it will work when there is an account with multiple sites? I think it will only work with 1 site id at a time. But when i have a fix we could try again.

frank8the9tank, 2022-08-30:
Looking at the code i see:

`    def requestListOfAllPanels(self):
        json_obj = json.loads(self.requestLogicalLayout())

        aantal_panelen = len(
            json_obj["logicalTree"]["children"][0]["children"][0]["children"]
        )
        # print(aantal_panelen)

        paneel_ids = []

        for paneel in json_obj["logicalTree"]["children"][0]["children"][0]["children"]:
            # print("id: {}".format(paneel["data"]["id"]))
            # print(paneel)
            # paneel_ids.append(paneel["data"]["id"])
            paneel_ids.append(
                "{}|{}".format(paneel["data"]["id"], paneel["data"]["serialNumber"])
            )

        return paneel_ids`

But this only looks at: json_obj["logicalTree"]["children"][0]["children"][0]["children"]:

My inverter has 3 strings: first sting is zero panels. seconds and thirt string is 19 panels each.

so is has to look at json_obj["logicalTree"]["children"][0]["children"][1]["children"] and json_obj["logicalTree"]["children"][0]["children"][2]["children"]:

could that be?

ProudElm, 2022-08-30:
Yeah, the solution is there some where ;-) I have a example JSON with 2 inverters so i'm able to test and change stuff..

frank8the9tank, 2022-08-30:
Do you need mine? i don't want to post my id and optimizer data here, where can i contact you directly?

ProudElm, 2022-08-30:
Yeah, would be good to test the solution. you can use the email adres from this account or DM if that exists.. :)

frank8the9tank, 2022-08-30:
your on tweakers right? i've send you a message there.

ProudElm, 2022-09-02:
I've updated the code, it should work now. Please let me know if it does.

I've tested the JSON you've send me. The code should work, but i could not test it in Home Assistant.

frank8the9tank, 2022-09-02:
I've updated the integration and indeed its showing all the optimizers,
So that's good.
There is no data for them yet, but i persume that has some thing to do with the 15 minute update interval,
Will keep an eye on that this afternoon.

ProudElm, 2022-09-02:
Nice! Yeah, the entities are added but not updated. this would slow down the adding process. So after about 15min data should start to appear!

frank8the9tank, 2022-09-02:
Yes, the data is there,
Great work!!!!

Is it also possible to get the kwh data per optimizer?
[image]
The data is there, 1468 kWh, in this example.

ProudElm, 2022-09-06:
I dont think i can. The data is there at the time of adding the panels. But not when i gather the actual data we would like to see. If we want the unscaledenergy from your example, this would mean a second request per panel. Im not sure if this would be a good practice. If you have eq 30 panels, this would be 60 requests per 15 min. Adding up to 5760 per day.. ;-) I think there was a max number, but im not sure..

spali, 2022-09-21:
> I dont think i can. The data is there at the time of adding the panels. But not when i gather the actual data we would like to see. If we want the unscaledenergy from your example, this would mean a [...]

I think it could be possible with one request per site.
When I go to the detail page in the monitoring and press the refesh button (rop right), I can see a call in the browser:
`https://monitoring.solaredge.com/solaredge-apigw/api/sites/${siteId}/layout/energy?timeUnit=DAY`
It has a payload of a list of ID's which somehow are connected with the panels.

I redacted the ID's to be sure.
The first ID is the my siteId.
The second has the same values as the first. (I assume it represents the inverter, probably on multi-inverter site setups this shows the values per inverter).
The rest are panel values. Some in Wh and some in kWh, but it's always stated in `units`.

The response looks like 
```json
{
  "1234567890": {
    "energy": 79.73,
    "moduleEnergy": null,
    "unscaledEnergy": null,
    "units": "kWh",
    "color": "#A0ACF2",
    "groupColor": null,
    "relayState": null,
    "cellularConnectionProperties": null
  },
  "1234567890": {
    "energy": 79.73,
    "moduleEnergy": 1.48,
    "unscaledEnergy": 79733,
    "units": "kWh",
    "color": "#30F291",
    "groupColor": "#3091F2",
    "relayState": null,
    "cellularConnectionProperties": {
      "connectionType": "STANDARD",
      "connectionFailure": null,
      "connectable": true
    }
  },
  "1234567890": {
    "energy": 340,
    "moduleEnergy": 340,
    "unscaledEnergy": 340,
    "units": "Wh",
    "color": "#050F19",
    "groupColor": "#000000",
    "relayState": null,
    "cellularConnectionProperties": null
  },
  "1234567890": {
    "energy": 2.61,
    "moduleEnergy": 2.61,
    "unscaledEnergy": 2606.5,
    "units": "kWh",
    "color": "#2D89E5",
    "groupColor": "#000000",
    "relayState": null,
    "cellularConnectionProperties": null
  },
...
}
```

ProudElm, 2022-09-22:
Ah i see yeah, that would be only one request total, not per..

If i have time, i will see if i can add this, could be nice yes.

ProudElm, 2022-10-08:
Not much, life stuff got in between.. I hope to pick this up later this week.

I found that for each sensor there is a request, i thought it would group them by device. So im trying to make is smarten and thus reducing the total number of requests. If this works, i could add a request/sensor for the total amount of energy generated.

ProudElm, 2022-10-14:
I've been working on this now, but i dont get the url working:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/SITEID/layout/energy?timeUnit=ALL

This will give me a 405 error. 

In HA you can add a template sensor for the panel kWh total. see:
https://www.home-assistant.io/integrations/integration

I will try to see if a can get acces to it, but if not, the above solution will work.

ProudElm, 2022-10-17:
[image]

Yes, finally. I got i working.. I needed to 'hack' that URL. It needed an extra information from a cookie that is present when viewing the information from the browser. It works for me now. I will be testing it some more.

frank8the9tank, 2022-10-17:
that looks very nice, When do you think you can release a new version?
I will check it on my side when you have added a new release.

ProudElm, 2022-10-17:
Not sure yet, hopefully tomorrow.

ProudElm, 2022-10-18:
I just released version 1.1.0! Let me know if you find any weird stuff if not, i will close this issue.

willem-ebbers, 2022-10-18:
Hi, 

I just installed the latest version and found the following error in my HA log

```
2022-10-18 12:48:35.864 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.lifetime_energy_1_1_15 fails
Traceback (most recent call last):
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
    raise JSONDecodeError("Expecting value", s, err.value) from None
  File "/usr/local/lib/python3.10/json/decoder.py", line 355, in raw_decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.10/json/decoder.py", line 337, in decode
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.10/json/__init__.py", line 346, in loads
    lifetimeenergy = json.loads(self._client.getLifeTimeEnergy())
  File "/config/custom_components/solaredgeoptimizers/sensor.py", line 168, in update
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    raise exc
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 710, in async_device_update
    await self.async_device_update()
```
all other entities update properly, only the lifetime energy entity stays unknown:
[image]

ProudElm, 2022-10-24:
The original issue has been solved.

closed 2022-10-24 by ProudElm