ProudElm/solaredgeoptimizers #4: Error while setting up solaredgeoptimizers: KeyError: 'Current [A]'

slyoldfox, 2022-09-11:
```
2022-09-11 21:55:28.609 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.1.1 to Home Assistant
2022-09-11 21:55:28.611 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.current_1_1_1
2022-09-11 21:55:28.612 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.optimizer_voltage_1_1_1
2022-09-11 21:55:28.613 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.power_1_1_1
2022-09-11 21:55:28.613 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solaredgeoptimizers entity: sensor.voltage_1_1_1
2022-09-11 21:55:29.326 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solaredgeoptimizers platform for sensor
Traceback (most recent call last):
  File "/Users/slyoldfox/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
    await asyncio.shield(task)
  File "/Users/slyoldfox/.homeassistant/custom_components/solaredgeoptimizers/sensor.py", line 75, in async_setup_entry
    info = await hass.async_add_executor_job(
  File "/usr/local/Cellar/python@3.10/3.10.5/Frameworks/Python.framework/Versions/3.10/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/slyoldfox/.homeassistant/custom_components/solaredgeoptimizers/solaredgeoptimizers.py", line 54, in requestSystemData
    return SolarEdgeOptimizerData(itemId, decodeResult(r.text))
  File "/Users/slyoldfox/.homeassistant/custom_components/solaredgeoptimizers/solaredgeoptimizers.py", line 204, in __init__
    self.current = json_object["measurements"]["Current [A]"]
KeyError: 'Current [A]'
```

I have 8 panels. I expected I had 8 optimizers, but it seems the logical layout is showing 16. 8 of them have a `lastMeasurementDate`- the other don't.

I made some modifications on the code base and might push a PR in the coming days.

ProudElm, 2022-09-12:
Oh. Dit you at any time had more then 8? Or where they replaced at some point? If you are able to check the JSON response from: 
https://monitoring.solaredge.com/solaredge-apigw/api/<<SITEID>>/sites//layout/logical
(change <<SITEID>> to your site id)

Then i think it could be an issue that there are optimizers listed but not active? I dont check for that hence there are 16 found?

slyoldfox, 2022-09-12:
They were placed last week and I only paid for 8 haha.

i got some code ready that works from my side, I’ll push a PR this evening.

slyoldfox, 2022-09-12:
See #5

closed 2022-09-13 by ProudElm