ProudElm/solaredgeoptimizers #6: Sensor names

Giel538, 2022-09-13:
Hi,

Is is possible that all sensor names are a bit more specific. 

For example: sensor.current, sensor.current_2, etc.

Is it possible to get something like this: sensor.optimizer1_1_1_current, sensor.optimizer1_1_2_current, etc.

If i have for example allready a sensor.current and i load this plugin i got for my first panel sensor.voltage, sensor.power and for current: sensor.current_1

ProudElm, 2022-09-14:
In the next version this should be fixed. But i will check to make sure.

Giel538, 2022-09-14:
I see version first in HA but i cannot update. Nothing happands after pressing update button

ProudElm, 2022-09-15:
I know, i messed something up with this tag/release.. 👎

ProudElm, 2022-09-15:
Laters release works again in HACS and uses the now better naming.

closed 2022-09-15 by ProudElm

Giel538, 2022-09-15:
still not been able to update.

Giel538, 2022-09-15:
after some deleting and re-adding i am been able to update from 1.03 to 1.04

sensor names did not change or do i need to re-add the integration? or wait longer ?

ProudElm, 2022-09-16:
Ok thanx! i was not sure if the names would auto-update. But i guess they don't.  Now we know for sure. Thank you!

Good point, i will look into that ;-) Thanx!