ProudElm/solaredgeoptimizers #9: Integration stops updating

rtfmjoey, 2022-09-27:
After recently updating to the new version which supports multiple inverters I noticed that this version stops updating after some time.
Only resolution seems to remove and re-add the integration, restarting/rebooting HASS doesn't solve the issue.

When re-adding, the integration seems to fail after 1 or 2 days and re-adding solves the issue again for 1-2 days.

The logs show the following:

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:796
Integration: Sensor (documentation, issues)
First occurred: 26 september 2022 om 13:34:13 (87 occurrences)
Last logged: 11:04:13

Updating solaredgeoptimizers sensor took longer than the scheduled update interval 0:15:00

The SE monitoring portal shows recent data, so the source doesn't seem to be the issue. 

I've used Domoticz until I discovered this integration to monitor the optimizers and when I switch back to Domoticz the optimizer data is pulled just fine.

I'm currently running HASS version 9.8.6 and have no issues with other integrations.

ProudElm, 2022-09-29:
Ai.. That's not good.. I cant see why, its been running fine for days here. But i found out that in stead of 1 call per device to update the 4 sensors. it makes a call for every sensor. So instead of 1 request per panel there are 4. Perhaps this is causing issues at your end?
I'm trying to rebuild some part to prevent this from happening.

rtfmjoey, 2022-09-29:
I can't tell if thats the issue, I do know that SE limits HTTP requests to once every 15 minutes or something like that.

HASS had lots of updates the last few weeks and I've been installing them right away thus causing a restart of HASS. 
It seems that the integration stops when HASS is restarted. I haven't restarted HASS the last few days and the integration is still running.

I'll make sure HASS doesn't restart for the next few days and will check if the integration keeps updating.

ProudElm, 2022-09-29:
I do have it build in such a way that the intergration wont update on startup etc. Because this would slow it all down. Perhaps that is interfering with updating/restarting HASS? I will try to test some my self.

rtfmjoey, 2022-10-03:
I've tested the last few days with restating HASS, updating HASS, even crashed HASS and restarted and the integration runs just fine.

At this point it's really uncelar why the updating stopped, perhaps it was a SE portal issue?

rtfmjoey, 2022-10-12:
Must be, integration has been running without issues for the last 2 weeks...  I'll close this issue.

closed 2022-10-12 by rtfmjoey