ProudElm/solaredgeoptimizers #10: Error with life tome energy sensor.

ProudElm, 2022-10-18:
Hi, 

I just installed the latest version and found the following error in my HA log

```
2022-10-18 12:48:35.864 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.lifetime_energy_1_1_15 fails
Traceback (most recent call last):
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
    raise JSONDecodeError("Expecting value", s, err.value) from None
  File "/usr/local/lib/python3.10/json/decoder.py", line 355, in raw_decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.10/json/decoder.py", line 337, in decode
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.10/json/__init__.py", line 346, in loads
    lifetimeenergy = json.loads(self._client.getLifeTimeEnergy())
  File "/config/custom_components/solaredgeoptimizers/sensor.py", line 168, in update
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    raise exc
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 710, in async_device_update
    await self.async_device_update()
```
all other entities update properly, only the lifetime energy entity stays unknown:
[image]

_Originally posted by @willem-ebbers in https://github.com/ProudElm/solaredgeoptimizers/issues/2#issuecomment-1282218256_

ProudElm, 2022-10-18:
uh um, weird..  Looks like you're not getting an json response.. 
The point was, it was a bit tricky to get this.

I'm doing 2 webcalls for it, 1 to get a cookie en then using some information from this cookie to make a call to:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/SITEID/layout/energy?timeUnit=ALL

Perhaps that is not working for you? I will try to add come logging to check what kind of response there is.

frank8the9tank, 2022-10-18:
I have a similar thing in in my logs:
`This error originated from a custom integration.

Logger: homeassistant.helpers.entity
Source: custom_components/solaredgeoptimizers/sensor.py:168
Integration: SolarEdge Optimizers Data (documentation, issues)
First occurred: 14:18:46 (76 occurrences)
Last logged: 14:35:30

Update for sensor.lifetime_energy_1_2_15 fails
Update for sensor.lifetime_energy_1_2_16 fails
Update for sensor.lifetime_energy_1_2_17 fails
Update for sensor.lifetime_energy_1_2_18 fails
Update for sensor.lifetime_energy_1_2_19 fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 515, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 710, in async_device_update
    raise exc
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solaredgeoptimizers/sensor.py", line 168, in update
    lifetimeenergy = json.loads(self._client.getLifeTimeEnergy())
  File "/usr/local/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.10/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
`

no data:
[image]

frank8the9tank, 2022-10-18:
The other 4 entities are updating, so those are still working
its only the lifetime energy thats not working.

willem-ebbers, 2022-10-18:
in the solaredgeoptimizers.py I do see that the SITEID is fixed to 1871534 (your site ID?) on lines 89, 106, 114
could that be contributing to the issue?

ProudElm, 2022-10-18:
Oh no.. thats a mistake indeed.. I will fix this.. That's my siteid indeed.. But that is also used to login, so  yeah, could be the problem.

ProudElm, 2022-10-18:
I fixed this in the latest version.. Please let me know if it works

willem-ebbers, 2022-10-18:
This works as a charm thanks!

ProudElm, 2022-10-18:
That's good to read!

ProudElm, 2022-10-24:
Life time sensors are now working

closed 2022-10-24 by ProudElm