ProudElm/solaredgeoptimizers #12: Lifetime energy sensor fluctuates

B1rdEater, 2022-10-24:
First of all; thanks a lot for you beautiful integration! 

Some panels report a drop in life-time energy production, and will continue on the normal level a few hours later. I use the life-time energy sensor to calculate to daily energy production per panel (and this really messes it up). Which panels show this behaviour is different everyday, also the number of panels is different. Today 6 out of 9 have a problem. Yesterday they were all correct.

Any idea if this can be fixed?

Wrong:
[image]

Correct:
[image]

ProudElm, 2022-10-24:
This is weird. I just checked my sensor and it has the same drop-off.. Not sure what could be causing this yet. I will take a look.

B1rdEater, 2022-10-24:
Thanks for investigating!

B1rdEater, 2022-10-25:
Today:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Henri/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Henri/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Panel/Date | 25-10-2022
-- | --
1.0.1 | X
1.0.2 | X
1.0.3 | X
1.0.4 | X
1.0.5 | X
1.0.6 | X
1.0.7 | X
1.0.8 | V
1.0.9 | V

</body>

</html>

B1rdEater, 2022-10-25:
Today's score. I will keep track; see if there is a clue.

[image]

B1rdEater, 2022-10-25:
Yeah, there is a really high chance that the problem is at SolarEdge its side. If so, can we submit a bug there?
Would be great if somehow you would be able to build a logic that prevents this drop with faulty data

ProudElm, 2022-10-25:
Uhm wel.. i kind of 'hacked' the data from the portal site. So if a bug report is a good idea..? :)
Kind of looks like a recalculation off the value of the optimizers when a new day happens.. It alway just a few wh..

I'm looking into if a round the value if this issue can be solved.

B1rdEater, 2022-10-25:
Whaha; well, than our best option won't be to tell SE about your intrusion. And indeed, it it just a few wh, but it messes up the utility meters big time, they will not start at 0.

[image]

B1rdEater, 2022-10-31:
Small update (was away for a few days). Affected panels seem to very A lot..

[image]

ProudElm, 2022-11-01:
Yes, i cant find a reason why.

I can solve it by checking the new value against the current one (i think) and only allow a new value if greater then the old value.
I hoped not to intervene to much in de data received, but it seems there is no other option. 

I will try to release a fix for it this week.

B1rdEater, 2022-11-01:
Sounds like a plan. Thanks for this fix; I will test it when you release the update

ProudElm, 2022-11-08:
With the latest release (v1.1.2) this wont happen again.

B1rdEater, 2022-11-09:
Seems to fail :(

[image]

[image]

ProudElm, 2022-11-09:
ai, that's weird... it should not update the sensor when the value is lower then its previous state.

Let me know if it happens again tomorrow, in the mean time, i will debug some more.

B1rdEater, 2022-11-09:
Yeah, its strange. I will check again tomorrow!

B1rdEater, 2022-11-10:
So far so good! How strange it didn't work yesterday. I will keep checking for a few days :)

ProudElm, 2022-11-10:
I found the problem... It was a stupid.. Next release it will be fixed (again)

B1rdEater, 2022-11-10:
It happens :). Thanks!

B1rdEater, 2022-11-10:
I am afraid that update somehow broke some other sensors, no new update for te last 2 hours:

[image]

[image]

[image]

[image]

Only lifetime-sensor keeps sending updates:

[image]

B1rdEater, 2022-11-11:
Seems alright now; all sensors and no dip in lifetime :D

ProudElm, 2022-11-11:
Nice! I will keep this issue open for a few more day's, just to be sure its really fixed ;-)

B1rdEater, 2022-11-11:
Smart move. Lets hope I get some more sun.. because its grey as hell at the moment here xD

B1rdEater, 2022-11-13:
Looking good! (inside blue square= SolarEdge; other = Enphase).

[image]

ProudElm, 2022-11-25:
@B1rdEater How is the data performing? Is it still stable?

B1rdEater, 2022-11-25:
Daily lifetime is functioning as expected and stable.

Not sure what happened yesterday with other sensors; but they were offline for some time. At the moment they are working as they should. See history. Red: power outages. Blue: can't explain why data was not available.

[image]

ProudElm, 2022-11-25:
nice to hear.

It's just a bit tricky some times. I want to refactor the way the data is retrieved, but i need to find out how ;-)

B1rdEater, 2022-11-25:
I understand! I think this issue can be marked as solved now :)

B1rdEater, 2022-11-25:
Mm, now I look at it again my sensors have been frozen for the past 3 hours (except lifetime).

The response was: {"194373381":{"energy":291.07,"moduleEnergy":291.07,"unscaledEnergy":291066.75,"units":"kWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373380":{"energy":280.95,"moduleEnergy":280.95,"unscaledEnergy":280954.0,"units":"kWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373383":{"energy":291.55,"moduleEnergy":291.55,"unscaledEnergy":291546.75,"units":"kWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373382":{"energy":276.41,"moduleEnergy":276.41,"unscaledEnergy":276412.5,"units":"kWh","color":"#2D89E5","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"196662828":{"energy":2.52,"moduleEnergy":0.28,"unscaledEnergy":2517418.0,"units":"MWh","color":"#30F291","groupColor":"#3091F2","relayState":null,"cellularConnectionProperties":{"connectionType":"STANDARD","connectionFailure":null,"connectable":true}},"194373379":{"energy":2.52,"moduleEnergy":0.28,"unscaledEnergy":2523602.0,"units":"MWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":{"connectionType":"STANDARD","connectionFailure":null,"connectable":true}},"194373388":{"energy":280.11,"moduleEnergy":280.11,"unscaledEnergy":280106.0,"units":"kWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373385":{"energy":281.27,"moduleEnergy":281.27,"unscaledEnergy":281272.5,"units":"kWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373384":{"energy":286.23,"moduleEnergy":286.23,"unscaledEnergy":286229.75,"units":"kWh","color":"#3091F2","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373387":{"energy":270.88,"moduleEnergy":270.88,"unscaledEnergy":270877.75,"units":"kWh","color":"#2D89E5","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null},"194373386":{"energy":265.14,"moduleEnergy":265.14,"unscaledEnergy":265136.0,"units":"kWh","color":"#2D89E5","groupColor":"#000000","relayState":null,"cellularConnectionProperties":null}}

B1rdEater, 2022-11-25:
Also, this freezing creates the same old problem. Sorry, I was to quick :(

[image]

B1rdEater, 2022-11-25:
Logger: custom_components.solaredgeoptimizers.sensor
Source: custom_components/solaredgeoptimizers/sensor.py:201
Integration: SolarEdge Optimizers Data ([documentation](https://github.com/ProudElm/solaredgeoptimizers), [issues](https://github.com/ProudElm/solaredgeoptimizers/issues))
First occurred: 24 november 2022 om 02:47:17 (15 occurrences)
Last logged: 13:49:42

'<=' not supported between instances of 'str' and 'float'
'<=' not supported between instances of 'float' and 'str'
('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))

B1rdEater, 2022-11-25:
As expected; just started today.... have a look at all daily yield graphs from the past 3 days. Today.. is kind of broken

[image]

ProudElm, 2022-11-25:
mmm... strange... does your system restart at 0.00? Causing those sharp declines perhaps?

Back to the drawing board i gues it is..

B1rdEater, 2022-11-26:
There is a lot of errors going on, that I haven't seen before. Is the code maybe broken somehow?

[image]

ProudElm, 2022-11-28:
not sure.. i'm seeing a lot of errors too.. I will try to make time this week to look into it!

B1rdEater, 2022-11-28:
Hopefully it is not too serious :(

B1rdEater, 2022-12-17:
It seems its officially dead now :(

[image]

ProudElm, 2022-12-17:
even after a reboot? I found one bug (not fixed) that is causing it to stop sometimes.. but a reboot solves this (temporally).

B1rdEater, 2022-12-17:
Yes that works!

ProudElm, 2023-01-29:
With the latest release it should me way more stable now, Can you let met know if it is indeed a lot better now?

B1rdEater, 2023-01-29:
Sure! Waiting for HACS to pick-up the latest release first :)

B1rdEater, 2023-02-04:
Updated last week and haven't spotted any error in the logging since. Looking good mate :)

B1rdEater, 2023-02-05:
[image]

Shit :(

ProudElm, 2023-02-05:
uhm. please help me, what's wrong? I think way more then your expecting? What sensor are you displaying there?

B1rdEater, 2023-02-05:
Its the lifetime sensor of some panels that is again dipping (issue we had before, see image below).

[image]

ProudElm, 2023-02-05:
ai.. i will check.. I hope its an easy fix..

closed 2023-02-07 by ProudElm

B1rdEater, 2024-04-01:
@ProudElm : problem still exists, did you manage to have a look a this?

[image]

ProudElm, 2024-04-01:
no not yet. i've started working again on this integration a few days ago, but i hate my dev-env. It is not stable to get it working. i will keep trying

B1rdEater, 2024-04-01:
Good to hear you picked up on the work here :). Hope you succeed in fixing this!

B1rdEater, 2024-06-30:
Any updates yet? This issue is closed; but the problem still exists :(

B1rdEater, 2025-02-21:
HI ProudElm! Has been a few months; did you manage to have a look at this issue?