ProudElm/solaredgeoptimizers #13: Collecting data takes long time

mountaindude, 2022-11-25:
This might very well be an issue at SolarEdge (pretty likely actually..) but still wanted to report it, in case it's something that can be handled well/less well in this component.

The whole thing might just be due to the rather large (?) number of devices and entities. 
On the other hand, for those having multiple inverters there might be way more optimisers, I guess..

**Background**
Trying to collect data from a 72 panel / 46 optimizer SolarEdge system. 

Running HA 2022.11.4 (=latest as of this writing)  in Docker on a rather beefy Linux server. 
Server performance shouldn't  be an issue thus.

**Symptoms/problem**
After installing the extension I see this in the log. 

[image]

Waiting for 25-30 minutes, then restarted the component.
Got this instead, but the 46 devices did show up now.

[image]

[image]

Each device is now sloooowly being. populated with lifetime energy, but that's it.
No other data makes it into these devices/entities. 

[image]

A bunch of errors in the logs though.

[image]

This one is the only one with a traceback:
[image]

Looks like most (all?) entities takes a long time updating:
[image]

All in all - the slow progress means that only a fraction of the entities have been updated during the 15 min sample interval.
Which might also cause problems (depends on how the code is written I guess), if a new extraction run starts even if the previous one hasn't finished yet.

Anyone else seeing this slow extraction of data?

mountaindude, 2022-11-25:
Btw, I have restarted the component to see if it was just a first/second-time-after-install issue, but no luck.
Same slowness.

ProudElm, 2022-11-25:
Ai.. Yeah, the way the integration works is kinda slow, so with such a big system, this would be an issue.  Sorry.

I don't now how to streamline it. The way it works it will request per device (a optimizer) the data from solaredge and than it will fill the sensors (not the lifetime).
This will take 10/15 sec normal per optimizer. But this will depend on network speed etc. 

The problem could be that the return file is so large that processing this each time is causing the problem. I don't know (yet) how i can only request the data once and then fill all the devices/sensors with 1 request.. This could solve many performance issues.

mountaindude, 2022-11-25:
No worries, I suspected something like that.
I never considered my SolarEdge system here particularly big thou.. But I guess we went a bit crazy on the optimisers :)

I might hack away on this too, to see if there's some other way of getting bulk data.
I am mainly interested in updates every few hours or even 1-2 times per day, maybe that makes the whole thing more feasible. We'll see. 

Really annoying that SolarEdge won't make panel-level data available via the APIs, but that's just how things are for now.

closed 2022-11-25 by mountaindude