ProudElm/solaredgeoptimizers #14: Total energy for panel

stepsolar, 2022-12-09:
E

ProudElm, 2022-12-09:
Cant you handle this not better via a template sensor? At least, if i understand you correctly? You want to enter the total amount of energy (kWh) produced per panel? Not the lifetime energy already provided but a custom number?

ProudElm, 2022-12-09:
You should have also a life time energy sensor. This is the total energy produced by the panel. 
[image]
This sensor should not reset per day but just keep on growing..

stepsolar, 2022-12-09:
Lifetime energy sensor is fine. But if it would be possible to insert an energy sensor of the day

ProudElm, 2022-12-09:
You can better use a utility meter from HA self. Use the life time sensor ass input and set the reset time to daily. Then you have a daily sensor. While your at it, use month/year too ;-)

stepsolar, 2022-12-09:
and what i am trying to do. Anyway thanks

stepsolar, 2022-12-10:
I can't get the same result shown in thesolardge app, could you help me thanks

stepsolar, 2022-12-11:
Greetings,
i almost managed to create the sensors but i wonder why in your project you didn't put the fetch of this data from the link "https://monitoring.solaredge.com/solaredge-apigw/api/sites/xxxxxx/layout/logical"
the color of the panel is also reported. You could create a card that changes color as shown in the Solardge screen

[image]

Thank you

ProudElm, 2022-12-11:
uhm never thought of it nor did i think i would use it..  There are still some issues with the code i need to sort out but dont have the time for.. those are my priority for now.

ProudElm, 2023-01-29:
Is this issue still valid or can i close it?

stepsolar, 2023-01-29:
.

closed 2023-01-29 by stepsolar