ProudElm/solaredgeoptimizers #15: counters stuck on value at sunset

Bassie007, 2022-12-14:
Hi, at sunset, all values of all individual panels remain stuck at a value (eg 1.04 w) and a new update will not be released until the next day
so it is difficult to recognize if there is nothing more generated by the respective panel
does something go wrong in my HA environment and should it just end on 0w /unknow because there will be no more update
or is this a feature by design?
[image]
[image]

ProudElm, 2022-12-15:
Hi, this is a problem caused by the solaredge portal. Once the inverter is no longer 'on' the data is no longer refreshed. Meaning the last value of the day will be sent even through the night untill the inverter gets started again and restart reporting values.

A possible solution i'm working on is that i'm checking the last update value and if this is longer then eq 1 hour i set all the relevant values to 0. But yeah, i'm still working on it..

Dutchy-79, 2023-01-10:
Same here, all values are stuck end of the day.
[image]

V1.1.3 installed

ProudElm, 2023-01-29:
I have resolved this issue by checking the last measurement value and if this is older then 1 hour, the sensors values are set to 0.

closed 2023-01-29 by ProudElm

Bassie007, 2023-01-31:
great job 👍