ProudElm/solaredgeoptimizers #16: entities not shown

Subia75, 2022-12-15:
I need some help
I wanted to try to load your integration to see the production of the panels but it doesn't create the entities for me

[image]

also i get this error

[image]

I tried to reload the integration but nothing.
I also installed the "home-assistant-solaredge-modbus" integration and it's been working fine for quite some time.
[image]

I don't want this integration to be the problem?
Thanks

ProudElm, 2022-12-16:
Can you email the response you are getting from the url (should be a json string):
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<<YOUR_SITE_ID>>/layout/logical (change the part with your site_id).
Perhaps your layout is somehow not wat the code is expecting..?

Subia75, 2022-12-16:
Thanks for the reply,
by default it asks to install version 1.1.3 which gives me problems, I tried to install version 1.1.4 and it created the sensors for me.
I don't know how.

ProudElm, 2022-12-17:
O weird that 1.1.3 is the default, that version had a bug indeed related to your problem.. Strange.. I will look into it.

Subia75, 2022-12-17:
I just took a screenshot and as you can see it offers me 1.1.3 as an update, it's actually weird.

 #
[image]

ProudElm, 2022-12-18:
oh weird.. i will check this asap.