ProudElm/solaredgeoptimizers #18: Not all panels found

freijn, 2022-12-16:
Many thanks for your work !  Really happy with it but....

The number panels found is stuck at 19 where I have 22.
Its running for 3 days now.

Any clue / hint?

THanks in advance.

     Frank

ProudElm, 2022-12-17:
What version is installed? It looks to default to version 1.1.3 while 1.1.4 is the most recent one?

spady7, 2022-12-20:
Hi, I' ve same issue. loaded 23 panels, but i've 28.
Installed v. 1.1.3 (compared to v. 1.1.4 there is not differences)
Any idea?

freijn, 2022-12-20:
After a few days, finnally all mu 22 panels are discovered. I Will see if i
can find anything in the logfiles.
Perhaps a small note in the install guide might be helpfull.

ProudElm, 2022-12-20:
mm...strange that after a time it does work.. dit you reboot perhaps?

@spady7 What could help is rebooting but if you can, could you send me the json response from:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<<YOUR_SITE_ID>>/layout/logical (change the part with your site_id).

You can email it to me. Than i can debug the response and check if there are some issues that could be causing not adding all the panels.

freijn, 2022-12-20:
Not reboot, but reload, after the update and yaml change. But the first 4
appeared a few min after implementatie. Then nothing and then after a few
days all have been found. I am remote but Will search the logfile for
details.

spady7, 2022-12-20:
> mm...strange that after a time it does work.. dit you reboot perhaps?
>
> @spady7 What could help is rebooting but if you can, could you send me the json response from: [...]

@ProudElm now seams to be ok. It takes around 30 min to get all data panels e panels too. Now seams to be ok.

freijn, 2022-12-20:
> You can email it to me. Than i can debug the response and check if there are some issues that could be causing not adding all the panels.

Can you let me know how I can find your email address? pls

ProudElm, 2022-12-21:
You can mail me at proudelm@yahoo.com

ProudElm, 2023-01-29:
@freijn Is this issue still active? If not, i will close it!

freijn, 2023-01-29:
After some days it did find all my panels.  You can close this issue.
Many thanks for your patience.

closed 2023-01-29 by freijn

ProudElm, 2023-01-29:
Good to read! Your welcome!