ProudElm/solaredgeoptimizers #19: [NewFeature] Is it possible to add sensor for actual energy of panel?

majkers, 2022-12-29:
Hi,
is it possible to add sensor for actually produced energy? I think it is available in this endpoint https://monitoring.solaredge.com/solaredge-apigw/api/sites/<<SITE_ID>>/layout/logical in reportersData sesction:

[image]

For now your integration have only lifetimeEnergy sensor

[image]

ProudElm, 2022-12-29:
No, i don't think this would be a good idea. (This would require an extra request per panel). If you want a daily sensor you can use a utility meter with the lifetime sensor as input and the cycle to daily. Then you can have a sensor with the daily value.

majkers, 2022-12-29:
Not at all. This would require only one request. All 
optimizers data are available as array in the result of endpoint I suggested

ProudElm, 2022-12-29:
Ok true. It would require 1 extra request. But with a 16 panel setup, it is now build to make 16 request + 1 request for the life time energy sensor. I don't see the reason to add 1 more request to get the daily value when you get the same request with a utility meter.

majkers, 2022-12-29:
> it is not Daily value. It is current ie. at the moment of update value. Ideał for making charts or more. It changes during the day

ProudElm, 2022-12-29:
but that value is reset after each day i think? And it keeps counting up? nog down i assume?

But for now, no, i wont look at this. Im trying to rebuild this intergration in order to make it more stable, So my focus is on that. If it all works again and is more stable (trying to decrease the number of requests) i will take a look at this again.

majkers, 2022-12-30:
The value is 0 when there is no production so it is dark outside. So one might say that it is reset every day but to be more accurate it falls to 0 after sunset and start rising after sunrise.

B1rdEater, 2023-02-27:
As ProudElm correctly states; this can easily be done in HA through utility meters. Panels within red scare are SolarEdge
(currently there is a communications issue, therefore numbers are incorrectly for today).

[image]

majkers, 2023-02-28:
Yesy I know. But having for example 20 panels forces you to create 20 utility meters extra besides sensors from integration.

ProudElm, 2023-02-28:
For the total number of sensors in HA it doesn't matter where you create them, you will still need to have them. 20 meters for total energy (provided by the integration) and 20 utility meters reset per day (those you make yourself).
I would still suggest using the utility meter function. And while you are at it, you can make ones for monthly and yearly.

B1rdEater, 2023-02-28:
For monthly and yearly data I would advice sending your daily-values to InfluxDB (using Node-Red) and plotting graphs from this data with Grafana. Saves you a lot of sensors :)

majkers, 2023-02-28:
You don't have to create any extra entity for this. It can be added as a new atribute to the existing ones like any other. So no need to create any extra entitites just use the ones provided by integration.