ProudElm/solaredgeoptimizers #23: Error fetching SolarEdgeOptimizer data: 'AEST'

genesispc, 2023-02-05:
I am getting this error after updating to version 1.2.0

This error originated from a custom integration.

Logger: custom_components.solaredgeoptimizers.sensor
Source: helpers/update_coordinator.py:215
Integration: SolarEdge Optimizers Data (documentation, issues)
First occurred: 9:04:29 AM (1 occurrences)
Last logged: 9:04:29 AM

Error fetching SolarEdgeOptimizer data: 'AEST'

ProudElm, 2023-02-06:
Yep, i know the problem. There's an issue with timezones..

Im working on the issue. See:
https://github.com/ProudElm/packaging_solaredgeoptimizers/issues/1

closed 2023-02-06 by ProudElm

genesispc, 2023-02-08:
Thanks for the update. Version 1.2.1 works a treat.
Now to try and implement that dashboard that you had detailed in on of the 'issues' :-)

ProudElm, 2023-02-09:
Thank you, good to hear! Good