ProudElm/solaredgeoptimizers #30: why password and not API key?

ianfriend, 2023-02-25:
Why does this integration need the username and password and not use the API key? Presumably because the optimiser data is not available via the API and you're actually doing something along the lines of scraping the website using the logon details?

ProudElm, 2023-02-26:
Yes that's the reason. The normal API wont allow for data from the optimizers. I use the username and password to login and retrieve the data.

ianfriend, 2023-02-26:
Ok thanks for the reply. While I'm sure there's many that allreciate your efforts with this integration, I would ask you to consider what this means from a security point of view.  At the very least making it much more clear what users supplying their username and password really means, as in theory they are giving away access details to potentially change many settings on their installation (depending on their level of access). 

Thanks for your consideration

ProudElm, 2023-02-26:
I'm unsure if that is even possible, but like al things on the internet, you need to be always on your gard and only supply this information if you trust the software. 
Luckily this is opensource, so one can always take look ;-)

The main reason i build this integration is to get the information from the optimizers. I hope that one day the information is also available via the modbus. But until than, this is the only way i know (and thus build).

Menz01, 2023-02-26:
> Why does this integration need the username and password and not use the API key? Presumably because the optimiser data is not available via the API and you're actually doing something along the lines of scraping the website using the logon details?

i cant even find an API on mine. i just got it installed la few weeks ago and turned on last week. i do not see anywhere in the web page to generate an API?

ianfriend, 2023-02-26:
You 

> i cant even find an API on mine. i just got it installed la few weeks ago and turned on last week. i do not see anywhere in the web page to generate an API?

You need admin access on the solaredge monitoring website to be able to generate API keys. Ask your installer to make you a site admin 👍

Menz01, 2023-02-27:
> You need admin access on the solaredge monitoring website to be able to generate API keys. Ask your installer to make you a site admin 👍

the company that installed for me said they could not make me a site admin but did generate an API key for me though

ianfriend, 2023-02-27:
> > You need admin access on the solaredge monitoring website to be able to generate API keys. Ask your installer to make you a site admin 👍
> 
> the company that installed for me said they could not make me a site admin but did generate an API key for me though

Hmm they definitely "can" but they may not want to. Not sure why as it doesn't give you access to anything dangerous. My installer made me a site admin so I can now generate my own API keys.

As long as you have an API key you're all good.  Doesn't help with this integration mind 😉

ProudElm, 2023-02-28:
I think it is clear why we need a username/password instead of the API key. Hopefully one day the let us get the optimizer data via the local modbus and we even don't need to use the cloud to get our own data..

bubbaedc, 2023-03-11:
is the user name different than the login email address? auth always fails when trying to login in HA, site ID, login, pw, all are correct, I can login on the solaredge web site, but not in HA? any ideas?

TIA!

ProudElm, 2023-04-24:
i missed this one. Do you have admin acces? Not sure if needed. But the login should be the same.

billkater, 2023-07-18:
> is the user name different than the login email address? auth always fails when trying to login in HA, site ID, login, pw, all are correct, I can login on the solaredge web site, but not in HA? any ideas?
> 
> TIA!

I have the same problem. My username/password does not work when trying to add this to HA. I am using the same username/password as I use on the mobile app and to website. The site ID is pulled straight from the website.

ProudElm, 2023-07-18:
Strange. I will try to add some logging  perhaps there is an other issue what is causing the error. Because the login should be the same and the username/password is just passed straight trough to the connection made to the portal.

chadburrows12, 2023-07-22:
I am having the same "Invalid Authorisation" error on my instance with the correct details. Did you find anything with the extra logging?

tubbers, 2023-12-13:
i am having the same invalid auth issue. is there any update?

[image]

TheoX76, 2024-05-02:
> @tubbers any change?

I also have this issue. I don't know if i'm an admin user in the portal. Is there a possibility to check if i'm?

tubbers, 2024-05-07:
> @tubbers any change?

Still a problem

renaudallard, 2024-08-22:
This seems to be about the same issues as in https://github.com/ProudElm/solaredgeoptimizers/issues/57

renaudallard, 2024-08-22:
In the web interface https://monitoring.solaredge.com, I don't see any place to check for optimizers data. Where do you get the data from? Could this error be there because I cannot see the optimizers data on the web interface?