ProudElm/solaredgeoptimizers #31: Cannot install integration

csomyy, 2023-02-26:
Dear @ProudElm 

I was really happy to find yout integration on HA.
I was about to use it. I downloaded the repository using HACS (not added the custom repo becouse HACS said it is in the main). Got to the devices and add intergration but no luck just waiting since minutes.
Do you have any clue?

[image]

Thanks! csomy

DavidqStokes, 2023-02-26:
I also had to wait quite a while, but it eventually got to the configuration screen. Currently now waiting for the first values to populate while I start changing my dashboard layouts to accommodate this.
I'm looking forward to being able to calculate how much additional energy I'm able to generate due to having string optimisers installed (two strings of 8 panels).

Big thanks to @ProudElm for developing this

ProudElm, 2023-02-26:
It can take a lot of time to gather all the information. Especially for rather large systems.

@csomyy Is it all working now? Perhaps check the HA log file for any errors?

@DavidqStokes Your welcome :)

csomyy, 2023-02-26:
Thanks for the reply.
I cannot find anything in the logs. No entry found on "SolarEdge Optimizers" expression or any parts.
I didn't configure anything beforehand. As i understand i need to do that on installing the integration. I only added the HACS repository.

ProudElm, 2023-02-26:
IK need to update the readme. But the intergration just got added to the default HACS installation. So it is no longer needed to install it. It can now be installed directly from HACS by searching in HACS and then installing the intergratoin.
Once installed, you can add it to HA like any other intergration.

But if you where able to add your site-id etc, then it is installed OK. It is kind of weird that nothing happens but that there is also no logging. Dit you attempt to restart HA?

DavidqStokes, 2023-02-26:
If have been able to set up the integration using my information. (note for the readme - make it clear that you need to enter the six digit site ID number, not the site name).

However, it's been 4 hours now and no entities have been created

ProudElm, 2023-02-26:
strange.. and nothing in the logs? 

I'll try to update the readme today!

csomyy, 2023-02-26:
HA has been restarted. I think it must be something with the HACS install. I'm waiting patiently.

DavidqStokes, 2023-02-26:
I'm still not seeing any sensors, so turned on error logging and received the following errors. I hope this helps 

Logger: homeassistant.components.sensor
Source: custom_components/solaredgeoptimizers/sensor.py:97
Integration: Sensor (documentation, issues)
First occurred: 15:28:10 (2 occurrences)
Last logged: 17:42:16

Error while setting up solaredgeoptimizers platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 293, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solaredgeoptimizers/sensor.py", line 97, in async_setup_entry
    info = await hass.async_add_executor_job(
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.10/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 70, in requestSystemData
    raise Exception
Exception

ProudElm, 2023-02-26:
You are sure you are using the correct site-id, username and password? The error is regarding retrieving the actual data, so i would assume it is correct. If you are in a browser with an active session to the solaredge portal and you try this link, does it work?

https://monitoringpublic.solaredge.com/solaredge-web/p/publicSystemData?reporterId=<PANEL_ID>&type=panel&activeTab=0&fieldId=<SITE_ID>&isPublic=true&locale=en_US

You need to replace the <PANEL_ID> with one paneel id and the <SITE_ID> with your site id.

To get a panel id see this link:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<SITE_ID>/layout/logical
(Again change the <SITE_ID>. This returns a JSON string where you can find the panel_id's like. It is in the childIds or childeren.

csomyy, 2023-02-26:
I didn't set up a thing. Didn't see it. Where can I input my data? I only
added the repo in HACS and tried to add the integration in settings/devices.

ProudElm ***@***.***> ezt írta (időpont: 2023. febr. 26., V
19:32):

ProudElm, 2023-02-26:
Oh sorry, i misunderstood.. But stil, the error should be a result of the first link. Could you still try it?

The site-id, username and password are correct at that point, the first test is to make sure of this and this error is past this point.

csomyy, 2023-02-26:
I'm in! I got access.
For a 3rd restart. Strange. I got access and got 24 devices for 24
optimisers. I'm checking the contents!

Good Job Sir!

ProudElm ***@***.***> ezt írta (időpont: 2023. febr. 26., V,
20:06):

ProudElm, 2023-02-27:
Nice! Not sure why this happend.. In rare cases i had while building issues with the communication to the protal. Perhaps that was causing this... Just glad it works! Have fun!

closed 2023-02-28 by ProudElm