ProudElm/solaredgeoptimizers #32: Second system

scannie67, 2023-02-27:
I have two SolarEdge inverters. 
Site-id is not the same. 
Is it possible to add a second system?
I want to see the optimizer information from both systems.

ProudElm, 2023-02-27:
Im not sure what happens. I think it will work, but i did not prepared the naming of the sensors for this case. I will try to test it.. ;-)

ProudElm, 2023-02-27:
It should work, the sensors are getting a unique_id based on the serial number. 

Let me know if it worked!

scannie67, 2023-02-27:
Every inverter has it own api and site-id. 
Also the password is not the same. 
But I can only fill in one. 
There is no option for the second one.

ProudElm, 2023-02-27:
You can just add an other instance of the same integration. Just add it again in HA en fill in the other side-id. This should work i think. The main problem could be the sensor names, but as far as unique-id's, that is using the serial number of the panel, so it should be unique even when using 2 systems.

ProudElm, 2023-03-05:
@scannie67 Did it work?

scannie67, 2023-03-05:
I donn’t know. The SolarEdge inverter is defective at this moment. After repair I try again.

ProudElm, 2023-03-06:
Ai, i'm sorry to hear that. Good luck with the repairs.

closed 2023-04-24 by ProudElm