ProudElm/solaredgeoptimizers #33: Cannot Create Sensors

DavidqStokes, 2023-02-27:
I think I confused things yesterday by posting in a different issue - my apologies

My issue is that although I can successfully install the integration and set it up using my SolarEdge credentials (site ID, user name and password), I have no sensors being created. I am getting the following Home Assistant error log entry:

        Logger: homeassistant.components.sensor
        Source: custom_components/solaredgeoptimizers/sensor.py:97
        Integration: Sensor (documentation, issues)
        First occurred: 10:40:41 (1 occurrences)
        Last logged: 10:40:41
        
        Error while setting up solaredgeoptimizers platform for sensor
        Traceback (most recent call last):
          File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 293, in _async_setup_platform
            await asyncio.shield(task)
          File "/config/custom_components/solaredgeoptimizers/sensor.py", line 97, in async_setup_entry
            info = await hass.async_add_executor_job(
          File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
            result = self.fn(*self.args, **self.kwargs)
          File "/usr/local/lib/python3.10/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 70, in requestSystemData
            raise Exception
        Exception

My HomeAssistant versions are as follows:
        Home Assistant 2023.2.5
        Supervisor 2023.01.1
        Operating System 9.5
        Frontend 20230202.0 - latest

I am in the UK

When I am logged in to the SolarEdge portal via Chrome browser (on the same network as my HA instance), if I try the following URL (https://monitoring.solaredge.com/solaredge-apigw/api/sites/<SITE_ID>/layout/logical) with my site ID I receive what I assume the JSON string with what I assume is the correct information (it makes sense with respect to my site).

If I try (https://monitoringpublic.solaredge.com/solaredge-web/p/publicSystemData?reporterId=PANEL_ID>&type=panel&activeTab=0&fieldId=<SITE_ID>&isPublic=true&locale=en) with my site ID and one of the panel IDs I receive a separate formatted string containing lots of information which appears to be correct (i.e. panel number, manufacturer, voltage, last reading date/time etc)

If useful, I can forward the strings that are returned

Thanks again for the work on this integration

David

ProudElm, 2023-02-27:
yes, if you can, that should be helpful. The error it self is not forwarded to HA, my mistake. But i can debug the error and hopefully i can see what the real problem is.

DavidqStokes, 2023-02-27:
I've just logged in to the SolarEdge portal and they are reporting performance issues via a pop-up. The requests I tried earlier are not working reliably.

I will therefore wait a while before trying again and providing the responses

DavidqStokes, 2023-02-27:
I restarted the integration and my 16 sensors with 96 entities have just shown up.
I wonder if this is linked to issues at SolarEdge?

ProudElm, 2023-02-27:
Ah yes, that could be.. I noticed that sometimes it takes a very long time while other times its really fast. 
It could well be that they restrict access to their portal site with a max bandwidth or something.

It is now working OK?

DavidqStokes, 2023-02-27:
Currently the entities are showing as Unavailable, but I'll give it a while

ProudElm, 2023-02-27:
All off them? They should update every 15 min.

DavidqStokes, 2023-02-27:
I restarted the integration and 15 of the 16 entities now have values. Looking at the 'last measurement' entity shows that the data from each panel was received (by SolarEdge?) at a different time

DavidqStokes, 2023-02-27:
SolarEdge's implementation might be somewhat poor, but your integration is now working well. For the first time since they were installed, I can estimate how much additional generation my optimisers are providing.

Thanks for all the work on this

ProudElm, 2023-02-27:
This method relais heavenly on the portal of SolarEdge, i notice a lot of downtime or errors from their website.. Not much we can do about it unfortunate. 

Your welcome!

closed 2023-02-27 by ProudElm