ProudElm/solaredgeoptimizers #34: Error in log - this will fail in version 2023.3

godfreym29, 2023-02-27:
Logger: homeassistant.helpers.frame
Source: helpers/frame.py:77
First occurred: 15:46:23 (1 occurrences)
Last logged: 15:46:23

Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for solaredgeoptimizers using this method at custom_components/solaredgeoptimizers/__init__.py, line 40: hass.config_entries.async_setup_platforms(entry, PLATFORMS)

godfreym29, 2023-02-27:
Also getting this error

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:503
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 15:46:30 (42 occurrences)
Last logged: 15:47:01

Entity sensor.last_measurement_1_18_1 (<class 'custom_components.solaredgeoptimizers.sensor.SolarEdgeOptimizersSensor'>) is using state class 'measurement' which is impossible considering device class ('date') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.lifetime_energy_1_19_1 (<class 'custom_components.solaredgeoptimizers.sensor.SolarEdgeOptimizersSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.last_measurement_1_19_1 (<class 'custom_components.solaredgeoptimizers.sensor.SolarEdgeOptimizersSensor'>) is using state class 'measurement' which is impossible considering device class ('date') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.lifetime_energy_1_20_1 (<class 'custom_components.solaredgeoptimizers.sensor.SolarEdgeOptimizersSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
Entity sensor.last_measurement_1_20_1 (<class 'custom_components.solaredgeoptimizers.sensor.SolarEdgeOptimizersSensor'>) is using state class 'measurement' which is impossible considering device class ('date') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

ProudElm, 2023-02-27:
Thank you. I thought i fixed the issue regarding platforms.. I will take a look.

About the date sensor.. I could not get it to work 'normal'. Hence i choose to use None as a device class... Dit you install it as 'new' since 1.2.1? Or dit you upgrade from an older version?

godfreym29, 2023-02-27:
I installed it last night using HACS, so it should be the latest version.
In HACS it says v1.2.1

Regards
Godfrey

Moorhead Consulting Ltd is a company registered in England and Wales. Company number: 3641057.
Place of Registration: 8 Station Road, Settle, North Yorkshire, BD24 9AA. VAT number: 708 4169 30

This email contains confidential information and is for the exclusive use of the intended recipient. Any form of distribution, copying or use of the information contained in the email is strictly prohibited and may be unlawful. If the message is received by anyone other than the intended recipient, please return the message to the sender by replying to it and then delete the message from your computer.

ProudElm, 2023-02-27:
> 

could be.. i'll try to fix it tonight..

ProudElm, 2023-02-27:
My bad.. i forgot i fixed it but did nog release it yet... It is fixed with version v1.2.2 :)

godfreym29, 2023-02-27:
OK, thanks

Regards
Godfrey