ProudElm/solaredgeoptimizers #35: i can install the integration without issue but does not create any entities or devices

Menz01, 2023-02-27:
Hello,
    i have no issue installing the integration. it accepts my username and password and site id but after that it does not show me anything. if i look at it there are no devices and no entities

ProudElm, 2023-02-27:
Are there any errors in de logging in HA?

Menz01, 2023-02-27:
> Are there any errors in de logging in HA?
> 
> Sometimes it can take a while, there are issues with performance on the SolarEdge portal side

not seeing anything in the logs but i'll admin i am not sure i am looking correctly. How much "time" should i wait? it's been over 48 hours now since i installed it

ProudElm, 2023-02-28:
oh, that's to long.. 

Perhaps remove the integration and try again? And check the home assistant logs for any errors or issues. Any error should be logged there.

Menz01, 2023-02-28:
I have removed the integration and re-added it a few times. never an error on the main screen but nothing shows after its added. i even put in my username wrong just to make sure an error would show and it did for that. not seeing anything in the logs. Perhaps i am not looking in the correct spot? can you detail how to get to the logs i should be seeing so i can see if i am doing it right?

ProudElm, 2023-02-28:
The HA logs are located @ settings ->System -> Logs.
If the integration has any errors, those should be available there.

Menz01, 2023-03-01:
so i opened two browser windows... one to the logs are you outlined and the other to the integrations page. i then uninstalled your integration and then re-installed it and there were zero entries in the logs. nothing at all changed from the last entry in the log before i started. i even refreshed and did not see any entries related to your integration.

ProudElm, 2023-03-01:
With install you mean in HA self, thus entering the site-id etc?

Can you if you want please send me the JSON result of the following request? So i could debug it here?
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<SITE_ID>/layout/logical

Thanx!

Menz01, 2023-03-01:
yes... i go to setting then devices and services, click the add
integration, search for your integration and add it. when it asks i put in
the site id, username and password and click ok and the integration adds
but nothing is there.

how do i get the JSON you are asking for? i go to that link and all i get
is a black page with all sorts of code on it

Menz01, 2023-03-01:
i think i figured it out

Menz01, 2023-03-01:
> With install you mean in HA self, thus entering the site-id etc?
> 
> Can you if you want please send me the JSON result of the following request? So i could debug it here? https://monitoring.solaredge.com/solaredge-apigw/api/sites/<SITE_ID>/layout/logical
> 
> Thanx!
[logical.zip](https://github.com/ProudElm/solaredgeoptimizers/files/10864383/logical.zip)

ProudElm, 2023-03-02:
Thank you!

I see you have some thing like a production meter? I did not know this was even possible. 
I now know the problem, i need to rework some code to make sure this wont be an issue any more.

ubarni, 2023-03-03:
Looks i am having the same issue (no entities created) and added my logical json and a screenshot.

[image]

[logical_2686247.zip](https://github.com/ProudElm/solaredgeoptimizers/files/10877851/logical_2686247.zip)

ProudElm, 2023-03-03:
Thank you. I

ProudElm, 2023-03-04:
I seem to have solved it. I will test some more and try to publish the fix tomorrow.

it seems my original test case (me myself and i) was not valid enough for all the different possibilities out there ;-) Hopefully most cases are now solved

closed 2023-03-05 by ProudElm

ubarni, 2023-03-05:
Congratulations. Now my Optimizers were found 🥰 Thank You... 
[image]

ProudElm, 2023-03-05:
Nice! Your welcome!

Menz01, 2023-03-05:
mine are now being found too... wow so many of them!!! thanks

Menz01, 2023-03-05:
[image]