ProudElm/solaredgeoptimizers #36: Integration only finds 16 of 28 optimizers

2xdehelft, 2023-03-01:
Hi,

Great integration! Nice to have the solaredge optimizer data in Homeassistant. Strange thing is it only detects the first 16 optimizers of 28 in total.

Any idea why? Please let me know if I can help in any way.

ProudElm, 2023-03-01:
All the optimizers are on 1 inverter? 
No errors or warnings in the HA log?

2xdehelft, 2023-03-01:
Only this in logging : Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.

They are all on one inverter and one string, but different types of optimizers and panels (but they are compatible)

2xdehelft, 2023-03-01:
Nevermind.. did an extra reboot of HASS, now all 28 are visible...

Thanks for the support!

ProudElm, 2023-03-01:
Oh nice!

I'll need to add this to the read-me.. It can take some time or a restart sometimes.. :-(

closed 2023-03-01 by ProudElm