ProudElm/solaredgeoptimizers #37: Home assistant core update 2023.3.0

MarcBotman, 2023-03-02:
Seen this in the logs:
Logger: homeassistant.helpers.frame
Source: helpers/frame.py:77 
First occurred: 1 March 2023, 22:50:28 (1 occurrences) 
Last logged: 1 March 2023, 22:50:28

Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for solaredgeoptimizers using this method at custom_components/solaredgeoptimizers/__init__.py, line 40: hass.config_entries.async_setup_platforms(entry, PLATFORMS)

MarcBotman, 2023-03-02:
Sorry; I have just seen that this is solved in version 1.2.2......
I should have checked before submitting...
Will not do it again, promised...

ProudElm, 2023-03-02:
haha.. Dont worry!