ProudElm/solaredgeoptimizers #39: Installs, identified devices and entities but not polling information, checked again after 24 hours

cralwalker, 2023-03-03:
**Current** in Unavailable 
and the other metrics are Unknown

[image]

thank you for this initiative!

ProudElm, 2023-03-03:
Weird. No errors in the HA logging?  Did you try a restart?

cralwalker, 2023-03-03:
Yep, weird. I have restarted, deleted and reinstalled the integration. HA log examples;

> 2023-03-03 17:04:49.617 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error in updating updater
> 2023-03-03 17:04:49.621 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] 
> 2023-03-03 17:04:49.625 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error fetching SolarEdgeOptimizer data:

ProudElm, 2023-03-03:
Not sure, i think there is only one cloud.

Im working one some issues and will add while i'm fixing those add some more logging. There is an error but its not logged.

ProudElm, 2023-03-05:
With the latest version it should return some more error logging. Could you check if that is true?

cralwalker, 2023-03-05:
I updated to version 1.2.3, enabled debug logging and restarted Homeassistant.
In the /config/home-assistant.log the following was recorded a minute ago

> 2023-03-06 07:38:10.782 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
> 2023-03-06 07:38:54.824 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: solaredgeoptimizers, sensor.solaredgeoptimizers
> 2023-03-06 07:39:00.579 ERROR (MainThread) [homeassistant.components.sensor] Setup of platform solaredgeoptimizers is taking longer than 60 seconds. Startup will proceed without waiting any longer.
> 2023-03-06 07:39:14.241 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error in updating updater
> 2023-03-06 07:39:14.245 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] 
> 2023-03-06 07:39:14.249 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error fetching SolarEdgeOptimizer data: 
> 2023-03-06 07:40:24.330 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error in updating updater
> 2023-03-06 07:40:24.333 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] 

My setup on a Raspberry Pi 4B
Home Assistant 2023.3.1
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230302.0 - latest

ProudElm, 2023-03-06:
Oh damm.. ik hoped the error would be visible with the latest version. 

Could you perhaps send me the response of:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<SITE_ID>/layout/logical

cralwalker, 2023-03-07:
Yep, before I do, perhaps you know the answer; how much of my site info can I share publicly before risking external interference by a bad actor? I had a visitor the other day taking a photo of one of my inverters' ID tag. I do not know this person at all and wonder what the risk is? Can I DM you the response?

ProudElm, 2023-03-07:
Uhm, thats weird.. I thought you can acces the internal accespoint of the inverter if you scan the ID-tag with set settapp from solaredge. At least, with the latest inverters. But you have to manualy switch that red switch. But im not sure... 

Yes, you can DM me the response!

DeFlanko, 2023-03-10:
I too have this same issue -- i can help with any troubleshooting if needed.

ProudElm, 2023-03-10:
I will try to push a release today where there is more logging so hopefully you will see the real error/reason why the integration is failing.

cralwalker, 2023-03-31:
Ping, if there are any other tests I can do to help?

ProudElm, 2023-03-31:
I was way too busy with work and private matters unfortunately. From this weekend it will be quieter and I will pick up some stuff again! Sorry for the inconvenience and yes if I need your help I will let you know.

DeFlanko, 2023-04-22:
@ProudElm any update?

DeFlanko, 2023-04-23:
Hey is it possible because the word PANEEL wasn't translated from NL to EN is why its broken? (i dont know how to exactly test this)

[image]

ProudElm, 2023-04-23:
No, i could not find any poblems. I'm on holliday so i could have more time. I Will take an other look.

DeFlanko, 2023-04-23:
adding Debug log -- not sure how helpful it it. Removed the site id. 

```log
2023-04-23 12:12:23.218 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solaredgeoptimizers which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-04-23 12:12:34.596 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Found all information for site: ######
2023-04-23 12:12:34.597 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Site has 1 inverters
2023-04-23 12:12:34.597 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers (48) found to Home Assistant
2023-04-23 12:12:34.597 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers from inverter: 1
2023-04-23 12:12:34.597 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.1 to Home Assistant
2023-04-23 12:12:35.601 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.2 to Home Assistant
2023-04-23 12:12:36.501 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.3 to Home Assistant
2023-04-23 12:12:37.395 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.4 to Home Assistant
2023-04-23 12:12:38.511 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.5 to Home Assistant
2023-04-23 12:12:39.528 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.6 to Home Assistant
2023-04-23 12:12:40.341 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.7 to Home Assistant
2023-04-23 12:12:41.372 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.8 to Home Assistant
2023-04-23 12:12:42.188 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.9 to Home Assistant
2023-04-23 12:12:43.125 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.10 to Home Assistant
2023-04-23 12:12:43.571 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
2023-04-23 12:12:44.053 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.11 to Home Assistant
2023-04-23 12:12:44.911 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.12 to Home Assistant
2023-04-23 12:12:45.748 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.13 to Home Assistant
2023-04-23 12:12:46.938 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.14 to Home Assistant
2023-04-23 12:12:47.816 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.15 to Home Assistant
2023-04-23 12:12:48.773 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.16 to Home Assistant
2023-04-23 12:12:49.901 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.17 to Home Assistant
2023-04-23 12:12:50.963 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.18 to Home Assistant
2023-04-23 12:12:51.850 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.19 to Home Assistant
2023-04-23 12:12:52.738 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.20 to Home Assistant
2023-04-23 12:12:53.614 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.21 to Home Assistant
2023-04-23 12:12:54.596 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.22 to Home Assistant
2023-04-23 12:12:55.738 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.23 to Home Assistant
2023-04-23 12:12:56.589 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.24 to Home Assistant
2023-04-23 12:12:57.556 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.25 to Home Assistant
2023-04-23 12:12:58.600 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.26 to Home Assistant
2023-04-23 12:12:58.600 DEBUG (MainThread) [custom_components.solaredgeoptimizers.sensor] Update vanuit de coordinator
2023-04-23 12:12:59.459 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.27 to Home Assistant
2023-04-23 12:13:00.382 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.28 to Home Assistant
2023-04-23 12:13:01.242 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.29 to Home Assistant
2023-04-23 12:13:02.211 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.0.30 to Home Assistant
[... cut ...]
2023-04-23 12:13:58.602 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error in updating updater
2023-04-23 12:13:58.603 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] 
2023-04-23 12:13:58.603 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error fetching SolarEdgeOptimizer data: 
2023-04-23 12:13:58.603 DEBUG (MainThread) [custom_components.solaredgeoptimizers.sensor] Finished fetching SolarEdgeOptimizer data in 60.003 seconds (success: False)
```

DeFlanko, 2023-04-23:
I think I found something. Changing the 60 to 300 on line 148 in the sensor.py allowed me to get results.

 ```python
 try:
            # Note: asyncio.TimeoutError and aiohttp.ClientError are already
            # handled by the data update coordinator.
            # changing form 60 to 300 to increase time out. 
            async with async_timeout.timeout(300):
                _LOGGER.debug("Update vanuit de coordinator")
                data = await self.hass.async_add_executor_job(
                    self.my_api.requestAllData
                )
```
 
[image]

cralwalker, 2023-04-23:
Here is my log from re-installation this morning.

> 2023-04-24 08:13:08.451 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up 
> solaredgeoptimizers platform for sensor
> Traceback (most recent call last):
>  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 304, in _async_setup_platform
>    await asyncio.shield(task)
>  File "/config/custom_components/solaredgeoptimizers/sensor.py", line 97, in async_setup_entry
>    info = await hass.async_add_executor_job(
>  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
>    result = self.fn(*self.args, **self.kwargs)
>  File "/usr/local/lib/python3.10/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 68, in requestSystemData
>    raise Exception
> Exception

cralwalker, 2023-04-25:
...and now it works for me. Thank you. 

Where do you discuss automations related to the optimizers, eg. sending a message if one panel suddenly drops in production indicating a hardware fault?

ProudElm, 2023-04-25:
Not sure. i really dont have a place for that. Perhaps make an issue and then we can do it there ;)

closed 2023-04-25 by ProudElm