ProudElm/solaredgeoptimizers #42: small change request

ChristophCaina, 2023-03-06:
Hi, just wanted to know, if the "last measurement" could be changed into "timestamp" device class, since it does not only provide a date, but also a time...

[image]

This would allow the sensor to be displayed similar to this:
[image]

rather than only showing a time stamp :)

ProudElm, 2023-03-06:
i've been struggling to get it usefully working. I parked the issue because of performance problems at the beginning. I will take a look at this again once i have some spare time!

ChristophCaina, 2023-03-06:
hm... I had a short look into it and payed around with some changes.

As far as I could see, changing the deviceClass to Timestamp alone does not work.
It's probably because the timestamp does not provide timezone information... ?

Typical timestamp format in HA:
`2023-03-06T05:42:00+00:00`

Information provided by the sensor:
`2023-03-06T14:13:19`

so, probably the +00:00 at the end needs to be added in order to work properly.

I will continue to play around with this and once I got it working, I will submit an PR :)

ProudElm, 2023-03-06:
Oh now i remember.. It has the do when comparing the time for if the value is to old to store i ran into problem with timezone's... 

I wont be fixing that.. I cannot get the correct timezone provided from the solaredge portal, so i skipped the hole timezone aware dates in python and that worked..

ChristophCaina, 2023-03-06:
ah okey :)

Right - with just using DeviceClass.Timestamp I am getting the following error:

```
    raise ValueError(
ValueError: Invalid datetime: sensor.last_measurement_1_1_1 provides state '2023-03-06 14:40:45', which is missing timezone information
```

Probably, the easiest way would be a template sensor which does set the DATE value as a timestamp.

I'll try to investigate deeper to see, if this can be changed at all.

ProudElm, 2023-03-06:
Yes i know.. i stopped for a reason ;-) There were bigger problems at the time..

template sensors fix everything in HA, always :-) hahah

closed 2024-08-02 by ChristophCaina