ProudElm/solaredgeoptimizers #45: Error: Connection aborted.', RemoteDisconnected('Remote end closed connection without response')

bbird8, 2023-04-04:
I recently get connection errors, indicated in the log as

`This error originated from a custom integration.

Logger: custom_components.solaredgeoptimizers.sensor
Source: custom_components/solaredgeoptimizers/sensor.py:178
Integration: SolarEdge Optimizers Data (documentation, issues)
First occurred: 08:01:23 (2 occurrences)
Last logged: 12:22:44

('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))`

followed by:

`This error originated from a custom integration.

Logger: custom_components.solaredgeoptimizers.sensor
Source: helpers/update_coordinator.py:182
Integration: SolarEdge Optimizers Data (documentation, issues)
First occurred: 08:01:23 (2 occurrences)
Last logged: 12:22:44

Error fetching SolarEdgeOptimizer data: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))`

Sometimes it's working after a few minutes again, sometimes not working until I restart HA.
Any idea?

---
Home Assistant 2023.3.6
Supervisor 2023.03.3
Operating System 9.5
Frontend 20230309.1 - latest

arjanhs, 2023-04-04:
Having the same issue.

Foxhill67, 2023-04-09:
Yep, same here. Home Assistant 2023.4.1 (Docker)

ProudElm, 2023-04-24:
i've had also the same issue. But i fear it has something to do at the other end, at solaredge. the web request fails because of a error/timeout at there end.
Only solution would be to make it that if a request fails, it will try again. I will try to see if a can make that, but it wont be any time soon, i'm now on holiday and cant acces my dev.

gjdv, 2023-04-24:
Have a look at https://github.com/gjdv/packaging_solaredgeoptimizers/commit/df94fc124ac08068038e36611309e49683c8f6be

ProudElm, 2023-07-06:
hi, i will try it out but yes, put in a PR :)

gjdv, 2023-07-06:
As per request: https://github.com/ProudElm/packaging_solaredgeoptimizers/pull/11

Doesn't cover all issues caused by the SE portal being instable, but as said, reduces problems a lot