ProudElm/solaredgeoptimizers #49: Large gaps of Data not being pulled

DeFlanko, 2023-04-25:
Once I initialized the integration with the new 300 second time out it was polling just fine then it just stopped for a large amount of time then briefly again and now not polling at all...

Here is a screenshot of one of the panels power entity: ```sensor.power1_1_25```
[image]

The logs:
```log
2023-04-23 16:45:06.553 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
2023-04-23 16:45:49.208 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: solaredgeoptimizers, sensor.solaredgeoptimizers
2023-04-23 17:37:42.000 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error in updating updater
2023-04-23 17:37:42.000 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Expecting value: line 1 column 1 (char 0)
2023-04-23 17:37:42.001 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error fetching SolarEdgeOptimizer data: Expecting value: line 1 column 1 (char 0)
2023-04-23 19:37:10.386 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error in updating updater
2023-04-23 19:37:10.388 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor]
2023-04-23 19:37:10.389 ERROR (MainThread) [custom_components.solaredgeoptimizers.sensor] Error fetching SolarEdgeOptimizer data:

```

ProudElm, 2023-07-06:
I have had very unstable moments, but at the same time, i could also not acces the portal it self.

How is it going now?

Vaszago, 2023-09-29:
I've had the query running for two weeks now and the last few days there have been very large gaps in the query. It was better before but there were rare but still some gaps here too. An SE problem or is it the AP.

[image]

Vaszago, 2023-09-29:
[image]

closed 2024-10-01 by DeFlanko