ProudElm/solaredgeoptimizers #51: Update frequenze

rogerthn2019, 2023-05-03:
Hi and thanks for a **greate** intergration!

You do say

> This intergration will update its sensors every 15 minutes. More frequent is not usefull because the portal will only update every 15 minutes.

My portal seems to be updated something like every 5:th minute and update const.py as below seems to increase "performance"
`UPDATE_DELAY = timedelta(minutes=5)`

I guess that lower than 5 isn't advisable ;-)
SolarEdge seems to have

> Daily Limitation
> Use of the monitoring server API is subject to a query limit of 300 requests for a specific account token and a parallel query limit
> of 300 requests for each specific site ID from the same source IP.

rogerthn2019, 2023-05-03:
Enclosing some stats from states table, the update and restart was done around 8.
No entries between midnight and 5 AM, do you know if the integration is "polling" 24/7 ?
[states.txt](https://github.com/ProudElm/solaredgeoptimizers/files/11382754/states.txt)

I'll leave at 5 minutes for a day or 2 and maybe go lower after that.

ProudElm, 2023-05-03:
The integration is polling, but it checks the last update date given by the panels. If that is longer then a certain time, it does not update the sensors.

Let me know if you retrieve any extra data points!

rogerthn2019, 2023-05-03:
> The integration is polling, but it checks the last update date given by the panels. If that is longer then a certain time, it does not update the sensors.

OK

> Let me know if you retrieve any extra data points!

Today there are 3584 entries with entity_id like 'sensor.last_measurement_1%', with UPDATE_DELAY = timedelta(minutes=5) since ~08:00
Yesterday with UPDATE_DELAY = timedelta(minutes=15) there where 1783
On 2023-05-01 there where 1688.
i.e.  UPDATE_DELAY = timedelta(minutes=5) do increase update frequenze

ProudElm, 2023-05-04:
ok. that could be, but is there any relevent extra data? Or is it more of the same? I wil l test it also when i'm back home.

rogerthn2019, 2023-05-05:
Power from panel 1 this morning, might not be exactly every 5th minute but less than 15 minutes between updates.
[image]

ProudElm, 2023-07-06:
Any updates on this? This the same? No problems with the increased updates?

rogerthn2019, 2023-07-06:
> Any updates on this? This the same? No problems with the increased updates?
No problems noted

ProudElm, 2023-07-07:
Thanx. I would like to make the time between updates configurable, but i need time to find out how :) For now, i will close this issue and hope i can implement this soon.

closed 2023-07-07 by ProudElm