ProudElm/solaredgeoptimizers #54: optimizer 2:1 or 1:1

Aikawa24, 2023-06-26:
hi, my P850 Optimisers are connected with 2 PV Modules, the SE monitoring shows it correctly, is there a way to get this 2:1 or 1:1 connection status as attribute?

ProudElm, 2023-07-06:
hi, im not sure what your saying. But you have 2 panels per optimizer? Not sure if this is possible. Could you DM me the response from this request:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<<YOUR_SITE_ID>>/layout/logical  (change the part with your site_id).

Thanx.

Aikawa24, 2023-07-06:
Hi, here is the Output

{"siteId":xxxxxxx,"expanded":true,"playback":true,"hasPhysical":true,"logicalTree":{"data":null,"numberOfChilds":0,"childIds":[112668316],"children":[{"data":{"id":112668316,"serialNumber":"xxxxxxx","name":"Wechselrichter 1","displayName":"1","relativeOrder":1,"type":"INVERTER_3PHASE","operationsKey":3},"numberOfChilds":0,"childIds":[125219845,114097686],"children":[{"data":{"id":114097686,"serialNumber":null,"name":"Strang 1.1","displayName":"1.1","relativeOrder":1,"type":"STRING","operationsKey":0},"numberOfChilds":0,"childIds":[112668337,112668336,239707461,112668327,112668326,112668325,112668324,112668335,112668334,112668333,112668332,112668331,112668330,112668329,112668328],"children":[{"data":{"id":112668330,"serialNumber":"119A8EB7-F0","name":"Modul 1.1.1","displayName":"1.1.1","relativeOrder":1,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668331,"serialNumber":"119AA541-91","name":"Modul 1.1.2","displayName":"1.1.2","relativeOrder":2,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":239707461,"serialNumber":"13DC5D1B-67","name":"Modul 1.1.3","displayName":"1.1.3","relativeOrder":3,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668329,"serialNumber":"119A9B1D-63","name":"Modul 1.1.4","displayName":"1.1.4","relativeOrder":4,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668327,"serialNumber":"119AA22E-7B","name":"Modul 1.1.5","displayName":"1.1.5","relativeOrder":5,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668324,"serialNumber":"119226C6-8F","name":"Modul 1.1.6","displayName":"1.1.6","relativeOrder":6,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668337,"serialNumber":"119AA16F-BB","name":"Modul 1.1.7","displayName":"1.1.7","relativeOrder":7,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668333,"serialNumber":"119A8B40-76","name":"Modul 1.1.8","displayName":"1.1.8","relativeOrder":8,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668325,"serialNumber":"119AA8DD-30","name":"Modul 1.1.9","displayName":"1.1.9","relativeOrder":9,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668335,"serialNumber":"119A9419-58","name":"Modul 1.1.10","displayName":"1.1.10","relativeOrder":10,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668336,"serialNumber":"119A8524-54","name":"Modul 1.1.11","displayName":"1.1.11","relativeOrder":11,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668326,"serialNumber":"119A8DC1-F9","name":"Modul 1.1.12","displayName":"1.1.12","relativeOrder":12,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668328,"serialNumber":"119ADEBB-44","name":"Modul 1.1.13","displayName":"1.1.13","relativeOrder":13,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668334,"serialNumber":"119A9512-52","name":"Modul 1.1.14","displayName":"1.1.14","relativeOrder":14,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":112668332,"serialNumber":"119A9417-56","name":"Modul 1.1.15","displayName":"1.1.15","relativeOrder":15,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]}]},{"data":{"id":125219845,"serialNumber":null,"name":"Strang 1.2","displayName":"1.2","relativeOrder":2,"type":"STRING","operationsKey":0},"numberOfChilds":0,"childIds":[125193142,125193143,125193141,125193855,125193146,125193144,125193145,125193862,125193863,125193861,125193859,125193857,125193864,125193865],"children":[{"data":{"id":125193859,"serialNumber":"119A99DC-20","name":"Modul 1.2.1","displayName":"1.2.1","relativeOrder":1,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193146,"serialNumber":"119ADED6-5F","name":"Modul 1.2.2","displayName":"1.2.2","relativeOrder":2,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193142,"serialNumber":"119A93B1-EF","name":"Modul 1.2.3","displayName":"1.2.3","relativeOrder":3,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193143,"serialNumber":"119A8E0B-44","name":"Modul 1.2.4","displayName":"1.2.4","relativeOrder":4,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193145,"serialNumber":"11990DEB-A2","name":"Modul 1.2.5","displayName":"1.2.5","relativeOrder":5,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193861,"serialNumber":"119A941F-5E","name":"Modul 1.2.6","displayName":"1.2.6","relativeOrder":6,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193864,"serialNumber":"119A9F06-50","name":"Modul 1.2.7","displayName":"1.2.7","relativeOrder":7,"type":"POWER_BOX","operationsKey":0},"numberOfChilds":0,"childIds":[],"children":[]},{"data":{"id":125193144,"serialNumb [... cut ...] L1","AC Voltage P2":"AC-Spannung - L2","AC Voltage P3":"AC-Spannung - L3","AC Current":"AC-Strom - L1","AC Current P2":"AC-Strom - L2","AC Current P3":"AC-Strom - L3","Total Reactive Power":"Blindleistung","DC Voltage":"DC-Spannung","Power Factor":"Leistungsfaktor"},"125193862":null,"125193863":null,"125219845":null,"125193861":null,"125193859":null,"125193857":null,"125193864":null,"125193865":null},"geminiOperations":{"DISCONNECT":"Trennen","CONNECT":"Verbinden"},"reporterAlertSummaryList":[]}

Aikawa24, 2023-07-06:
I see Energy and modulenergy, if ist equal ist one module connected to one optimizer
If its doubled than two modules are connected