ProudElm/solaredgeoptimizers #55: solaredgeoptimizers.py missing

RichieB2B, 2023-07-04:
On January 10th, 2023 in commit 7ea6b24ae578845b9fe8af4e717fa79bca7d2139 of v1.1.9 you removed `custom_components/solaredgeoptimizers/solaredgeoptimizers.py`

How this module can work without it puzzels me as it is now at v1.2.4 and `solaredgeoptimizers.py` is still imported in various places.

ProudElm, 2023-07-06:
It has been added to PyPi and this installed via pip install when installing the add-on.

RichieB2B, 2023-07-06:
The [version on PyPi ](https://pypi.org/project/solaredgeoptimizers/) is using another repo https://github.com/ProudElm/packaging_solaredgeoptimizers which still has `solaredgeoptimizers.py` but none of the improvements from this repo. How are these two repos related?

ProudElm, 2023-07-07:
This repo is the integration for in HA.
The orther repo (packaging_solaredgeoptimizers) is the python file for communication between the integration and the solaredge web portal. 

When i have a little more time, all improvements are added to either this repo or the other one. Depending on where the changes belong.

closed 2023-08-30 by ProudElm