ProudElm/solaredgeoptimizers #57: Site ID issues

RedBeard91, 2023-08-02:
I have API access enabled, however I'm getting invalid login each time. Is there any conflict if I have the official solar edge app installed?

ProudElm, 2023-08-30:
I would think not. You are sure you are using the correct login and password? I dont know why it could be different. But you are the second person with this problem. I'm still adressen more serious issues but i plan to build in some more logging to perhaps get the real error.

RedBeard91, 2023-08-30:
Well, I'm using the site ID right from the API on my solar edge profile and the user name and pass is the username and pass I use to get onto my solar edge profile. Is there any discrepancy there?

ProudElm, 2023-08-30:
mmm.. no that should work. No API key required. Just your site-id and the username/password you use to enter the solar edge portal.

The site id is a 7(?) digit number.

RedBeard91, 2023-08-30:
that's what I'm using, 7 digit site ID. The same ID number is found in the API access category under admin.

ProudElm, 2023-08-30:
strange... hopefully i can get the the other issue fixed so i can add some logging.

i will try to install the the other integration also, perhaps there is something there after all, but i wont think so.

jds11111, 2023-11-26:
My integration was working for over a year but somehow stopped. I deleted the integration, removed the HACS folder, then re-downloaded, and tried to reinstall the integration. It says unexpected error. I triple checked credentials and site ID. Here are the logs from HA:

```
This error originated from a custom integration.

Logger: custom_components.solaredgeoptimizers.config_flow
Source: custom_components/solaredgeoptimizers/config_flow.py:85
Integration: SolarEdge Optimizers Data (documentation, issues)
First occurred: 3:20:02 AM (2 occurrences)
Last logged: 3:20:56 AM

Unexpected exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/urllib3/util/connection.py", line 72, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/socket.py", line 962, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
socket.gaierror: [Errno -3] Try again

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 715, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 404, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 1058, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.11/site-packages/urllib3/connection.py", line 363, in connect
    self.sock = conn = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f5e782fdd50>: Failed to establish a new connection: [Errno -3] Try again

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/urllib3/connectionpool.py", line 799, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='monitoring.solaredge.com', port=443): Max retries exceeded with url: /solaredge-apigw/api/sites/2996688/layout/logical (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5e782fdd50>: Failed to establish a new connection: [Errno -3] Try again'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solaredgeoptimizers/config_flow.py", line 85, in async_step_user
    info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredgeoptimizers/config_flow.py", line 58, in validate_input
    if not await hub.authenticate(hass, data["username"], data["password"]):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solaredgeoptimizers/config_flow.py", line 44, in authenticate
    http_result_code = await hass.async_add_executor_job(api.check_login)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 24, in check_login
    r = requests.get(url, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='monitoring.solaredge.com', port=443): Max retries exceeded with url: /solaredge-apigw/api/sites/2996688/layout/logical (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5e782fdd50>: Failed to establish a new connection: [Errno -3] Try again'))

```

renaudallard, 2023-12-06:
I get the same invalid login. I am sure of my login/pass and site ID. Could it be something from solaredge preventing configuration?

ome911, 2024-05-25:
Also having an invalid authentication problem but can log in fine on the SolarEdge site. Site ID is correct, 7 digits and I have the HA cloud integration enabled in case that was the problem.

alebartos, 2025-03-13:
maybe the recaptcha verification?

Neocold, 2025-03-14:
> maybe the recaptcha verification?

This is what I am thinking, but the issue seems to be ignored.