ProudElm/solaredgeoptimizers #59: sensors missing after install integration and configuration without errors

apf-doit, 2023-08-30:
Hi, I can't see this integration working. I enabled debug logging just to find out there are no errors so far. The install has been taken place several hours before i pulled out the debug log but i can't find any of the sensors to work with (I replaced my site id with 0000000). What am I doing wrong?

[image]

```
Zeile    7: 2023-08-30 17:28:51.628 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solaredgeoptimizers which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
	Zeile  337: 2023-08-30 17:29:45.995 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
	Zeile 3312: 2023-08-30 17:57:40.933 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Found all information for site: 0000000
	Zeile 3313: 2023-08-30 17:57:40.933 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Site has 1 inverters
	Zeile 3314: 2023-08-30 17:57:40.933 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers (34) found to Home Assistant
	Zeile 3315: 2023-08-30 17:57:40.933 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers from inverter: 1
	Zeile 3316: 2023-08-30 17:57:40.933 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.1 to Home Assistant
	Zeile 3317: 2023-08-30 17:57:41.432 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.2 to Home Assistant
	Zeile 3318: 2023-08-30 17:57:41.981 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.3 to Home Assistant
	Zeile 3319: 2023-08-30 17:57:42.439 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.4 to Home Assistant
	Zeile 3320: 2023-08-30 17:57:43.128 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.5 to Home Assistant
	Zeile 3321: 2023-08-30 17:57:43.711 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.6 to Home Assistant
	Zeile 3322: 2023-08-30 17:57:44.166 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.7 to Home Assistant
	Zeile 3323: 2023-08-30 17:57:44.844 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.8 to Home Assistant
	Zeile 3324: 2023-08-30 17:57:45.530 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.9 to Home Assistant
	Zeile 3325: 2023-08-30 17:57:46.127 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.10 to Home Assistant
	Zeile 3326: 2023-08-30 17:57:46.623 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.11 to Home Assistant
	Zeile 3327: 2023-08-30 17:57:47.066 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.12 to Home Assistant
	Zeile 3328: 2023-08-30 17:57:47.609 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.13 to Home Assistant
	Zeile 3329: 2023-08-30 17:57:48.222 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.14 to Home Assistant
	Zeile 3338: 2023-08-30 17:57:48.702 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.15 to Home Assistant
	Zeile 3339: 2023-08-30 17:57:49.175 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.16 to Home Assistant
	Zeile 3340: 2023-08-30 17:57:49.822 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.17 to Home Assistant
	Zeile 3341: 2023-08-30 17:57:50.091 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
	Zeile 3342: 2023-08-30 17:57:50.316 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.18 to Home Assistant
	Zeile 3343: 2023-08-30 17:57:50.761 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.19 to Home Assistant
	Zeile 3344: 2023-08-30 17:57:51.265 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.20 to Home Assistant
	Zeile 3345: 2023-08-30 17:57:51.852 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.21 to Home Assistant
	Zeile 3346: 2023-08-30 17:57:52.327 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.22 to Home Assistant
	Zeile 3355: 2023-08-30 17:57:52.891 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.23 to Home Assistant
	Zeile 3356: 2023-08-30 17:57:53.367 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.24 to Home Assistant
	Zeile 3357: 2023-08-30 17:57:54.017 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.25 to Home Assistant
	Zeile 3358: 2023-08-30 17:57:54.486 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.26 to Home Assistant
	Zeile 3359: 2023-08-30 17:57:55.033 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.27 to Home Assistant
	Zeile 3360: 2023-08-30 17:57:55.656 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.28 to Home Assistant
	Zeile 3361: 2023-08-30 17:57:56.407 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.29 to Home Assistant
	Zeile 3362: 2023-08-30 17:57:56.841 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.30 to Home Assistant
	Zeile 3363: 2023-08-30 17:57:57.358 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.31 to Home Assistant
	Zeile 3365: 2023-08-30 17:57:57.754 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.32 to Home Assistant
	Zeile 3366: 2023-08-30 17:57:58.515 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.33 to Home Assistant
	Zeile 3367: 2023-08-30 17:57:58.891 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.2.34 to Home Assistant
	Zeile 3368: 2023-08-30 17:57:59.334 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Done adding all optimizers. Now adding sensors, this may take some time!
```

ProudElm, 2023-08-30:
mm.. Can you please share with me the json response from:
https://monitoring.solaredge.com/solaredge-apigw/api/sites/<<YOUR_SITE_ID>>/layout/logical (change the part with your site_id).

apf-doit, 2023-08-30:
Hi, thanks for responding, yes i have the json output but is there a way for a direct message? The response is full of serials and ID's I do not like to share public because I am not sure how sensitive the data is.

ProudElm, 2023-08-31:
I'm not sure, but you can email me: job_sande at yahoo.co.uk

apf-doit, 2023-09-01:
Magically the optimizers appeared after 2 days while reloading the integration several times. No issue anymore.

closed 2023-09-01 by apf-doit