ProudElm/solaredgeoptimizers #60: intermetitent errors getting values from optimizers

Thlerry, 2023-09-29:
It looks like there are intermintent errors getting values from optimizers, and values come back as 0.

I can see in the logs the following errors.
It it seems the retrieval fails sometimes and others times are OK (intermittently).
Full logs attached.

Home Assistant Core
Error fetching SolarEdgeOptimizer data: Expecting value: line 1 column 1 (char 0)
5:37:06 PM – (ERROR) SolarEdge Optimizers Data (custom integration) - message first occurred at 5:05:32 PM and shows up 2 times
Expecting value: line 1 column 1 (char 0)
5:37:06 PM – (ERROR) SolarEdge Optimizers Data (custom integration) - message first occurred at 5:05:32 PM and shows up 2 times
Error in updating updater
5:37:06 PM – (ERROR) SolarEdge Optimizers Data (custom integration) - message first occurred at 5:05:32 PM and shows up 2 times
[home-assistant_2023-09-29T08-04-44.929Z.log](https://github.com/ProudElm/solaredgeoptimizers/files/12762296/home-assistant_2023-09-29T08-04-44.929Z.log)

Thlerry, 2023-09-29:
It seems issue #56 and #58 seem similar to what I reported here.

ProudElm, 2023-09-29:
Yeah i think this is due to communication errors between the solaredge site and HA. But i'm not sure.

Problem is, my inverters stopped communicating with the solaredge portal, its been almost a month now. So my integration stopped working for me and we cant figure out why it stopped (the inverter and its connection to the solaredge portal)..

until i get it working here, i cant continue debugging etc  :-(

Thlerry, 2023-09-29:
OK, let me know if I can help with the investigation in the meantime, or send you more logs.

Vaszago, 2023-10-05:
The Solaredge API supports 300 queries per day.
You could design the queries so that they are queried dynamically between sunrise and sunset. This would significantly increase the level of detail. I tested it. When I restart the integration, new values are always displayed. Also every 5 minutes.

Thlerry, 2023-10-16:
This is the error log in the UI:
[image]

[image]

Thlerry, 2023-11-20:
[image]

FYI: Still happening, I implemented a graph to keep track of the panel generation and you can see it's missing 2 values.