ProudElm/solaredgeoptimizers #61: Another new feature request: AC voltage inverter info (importent)

askmurphy, 2023-10-16:
Would be nice if we could see the AC voltage of the inverter, important..  
If the voltage is above 253V, the inverter will switch off. This happens when it's very sunny and the powergrid is overloaded.
If you have this often, you could decide to buy batteries or whatever..  
You can see this in the SolarEdge monitoring website dashboard, but we want it on our Homeassistant panel too ;-)

Thlerry, 2023-11-20:
This is already available with the SolarEdge modbus integration.
it's not part of the optimisers.

I've implemented a dashboard showing the voltage on each phase (I have a 3 phase install), and I have gauge turning red, and alert generated if it goes above 253, 256 etc.
BTW a battery will not solved the high voltage issue, you would need to contact your electricity provider and ask them to address the issue on the grid.

[image]

ProudElm, 2024-03-28:
i agree with @Thlerry. The modbus intergration is way better for this if only it could also read the optimizers then it would be golden! 

btw, nice dashboard! ;-)

closed 2024-03-28 by ProudElm