ProudElm/solaredgeoptimizers #63: [Feature request] TOTAL power generation combined from all panels

fencing49, 2023-11-01:
could we get another sensor built into the integration that would add all of the values from all of the optimizers?

for me personally - my solar provider doesn't want to add me as a site admin so i have no way of getting data about my system into HA without web scraping like what this integration does - which is awesome by the way.

i'd just like to get an additional sensor that adds all power or power and voltages (though i'm not sure that info would be useful)

ProudElm, 2024-03-28:
Best use a template sensor in HA to get the total.

ian-morgan99, 2024-05-18:
Could you not do this with the other SolarEdge integration to the inverter rather than to the optimizers?
I use that for the totals and this for the individuals.