ProudElm/solaredgeoptimizers #68: Integration Times out when Polling Optimizers during Setup

kaijk, 2024-03-16:
The integration finds the inverter, knows that there are 30 panels, tries to get data for the first one, then times out doing so.  Occasionally, the process will complete, but then will quit working after a bit of time.  This used to be very solid. I'm on HA v2024.2.2 in Docker, and v3.3.4 of this integration.

My SolarEdge App sees the data and my credentials have not changed.

Here are the log data:
```
2024-03-10 19:44:00.992 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Found all information for site: [redacted]
2024-03-10 19:44:00.992 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Found all information for site: [redacted]
2024-03-10 19:44:00.992 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Site has 1 inverters
2024-03-10 19:44:00.992 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers (30) found to Home Assistant
2024-03-10 19:44:00.992 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Adding all optimizers from inverter: 1
2024-03-10 19:44:00.992 INFO (MainThread) [custom_components.solaredgeoptimizers.sensor] Added optimizer for panel_id: 1.1.1 to Home Assistant
2024-03-10 19:44:09.757 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
2024-03-10 19:44:56.201 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: solaredgeoptimizers, solaredgeoptimizers.sensor
2024-03-10 19:44:59.757 ERROR (MainThread) [homeassistant.components.sensor] Setup of platform solaredgeoptimizers is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2024-03-10 19:46:11.709 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/connection.py", line 95, in create_connection
    raise err
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/connection.py", line 85, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 715, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 404, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1058, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 363, in connect
    self.sock = conn = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 179, in _new_conn
    raise ConnectTimeoutError(
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPSConnection object at 0x7f5f63f73410>, 'Connection to monitoringpublic.solaredge.com timed out. (connect timeout=None)')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 799, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='monitoringpublic.solaredge.com', port=443): Max retries exceeded with url: /solaredge-web/p/publicSystemData?reporterId=[redacted]&type=panel&activeTab=0&fieldId=[redacted]&isPublic=true&locale=en_US (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f5f63f73410>, 'Connection to monitoringpublic.solaredge.com timed out. (connect timeout=None)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solaredgeoptimizers/sensor.py", line 97, in async_setup_entry
    info = await hass.async_add_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 54, in requestSystemData
    r = requests.get(url, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 507, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPSConnectionPool(host='monitoringpublic.solaredge.com', port=443): Max retries exceeded with url: /solaredge-web/p/publicSystemData?reporterId=[redacted]&type=panel&activeTab=0&fieldId=[redacted]&isPublic=true&locale=en_US (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f5f63f73410>, 'Connection to monitoringpublic.solaredge.com timed out. (connect timeout=None)'))

kaijk, 2024-03-16:
Well, restarted. It started working.  We'll see.

kaijk, 2024-03-26:
Reopened.  Much more often than not, the integration fails as reported in the first post.

ProudElm, 2024-03-28:
i will check this with my own integration. problem remains that the connection to the site it self is unstable at times.

Mariusthvdb, 2024-04-29:
we now see:
```
Waiting on integrations to complete setup: {('solaredgeoptimizers', 'c9265c192165a4faa889bda950f09d3e'): 3391978.916099002}
```

which might be the same but worded differently? Not sure. this is running 2024.5.0b2

btw @ProudElm would you please check the PR by BDraco? It might help streamlining the startup proces

kaijk, 2024-07-18:
I have whitelisted the following IP addresses and the integration loads and acquires panel level data. I will monitor this, but it seems to work. You may or may not have to add all of these.  The last one I added, and that finally got it working is: `167.94.174.42`.

```
167.94.174.46 comment "ManualWlist: SolarEdge Optimizer Panel data"
167.94.174.42 comment "ManualWlist: Solaredge Monitoring"
167.94.110.0/24 comment "ManualWlist: SolarEdge pool for panel data"
```

kaijk, 2024-07-20:
Nope, the endpoints seem to hop about.  Now, I've whitelisted all of `solaredge.com`.  We'll see if that works.