ProudElm/solaredgeoptimizers #70: Lifetime Energy Sensor sends incorrect data

andyx87, 2024-04-14:
Hi, 
I have a problem with the energy sensors. The have sometimes strage datas.

[image]

castrovini85, 2024-04-15:
Same thing happening here. Some sensors have that jump by the start of the day (maybe when it starts to generate?) and some sensors are just fine.
[image]

ian-morgan99, 2024-05-05:
Same here . Every now and again they will just spike and sometimes return a few hours later [image]

ian-morgan99, 2024-05-13:
What I find odd, is that the 'Lifetime energy' appears to do a sudden jump at 7.45 or thereabouts most mornings:

[image]

The same happens on most of my panels. I wonder if some 'End of day' correction is happening wherever they hold their servers?

Kap3, 2024-05-30:
Same problem:
[image]

ian-morgan99, 2024-06-28:
For those that are finding the problems with the lifetime energy sensor really annoying , I'm now using the a Hacs plug in to calculate the integral between the real time power settings . It's not 100% accurate but it's better than the lifetime sensor. 
Unfortunately , when there is a SolarEdge server problem ; it results in very low readings , and doesn't recover itself like the lifetime energy does.

Mariusthvdb, 2024-10-28:
why not use the core SolarEdge integration for the lifetime values?

this custom integration is of added value because of the individual optimizer readings, but for the the data this cc duplicates the core integration, best use core data.

jhuang0, 2024-10-29:
> why not use the core SolarEdge integration for the lifetime values?
> 
> this custom integration is of added value because of the individual optimizer readings, but for the regular data this cc duplicates the core integration, best use core data.

I was hoping to use the 'lifetime' value to look at daily output of a given panel and to check to see if there are issues with individual panels. My original implementation (feeding this data into the utility meter) won't work because of this issue.

Mariusthvdb, 2024-10-29:
a right, sorry, I didnt realize you meant lifetime _per panel._
as a matter of fact, I like that idea, and might add those myself to my panel overview

[image]

doncuco, 2025-01-08:
So is this pull @Mariusthvdb  mentioned a solution for the issue?
I am seeing this as well and it is annoying. 
Maybe someone could approve it?

And @Mariusthvdb would you mind sharing the code for this view?

szerwi, 2025-01-25:
I experience the same issue. Has anyone found solution for that?

ernestk84, 2025-07-09:
Is there any solution?

ernestk84, 2025-07-09:
Hello,

All understand, but I don't know what is the problo to read lifetime energy..

[image]

There is problem, that integration not read data after cca. 4 PM, and in the morning is error in data.

It has already happened that today's lifetime was smaller than yesterday's.

BR