ProudElm/solaredgeoptimizers #73: Integration stops pulling data from optimizers

olizafiro, 2024-06-26:
Hi, since a couple of days, the integration stops pulling data from the optimizers. Once reloaded, the data are collected again but what seemed to be an isolated occurence happened already a couple of times in a few hours. I didn't change any specific setup to HA neither install an integration that could have led to this behavior except the last update to 2024.6.4 that I installed recently. Thanks for advices and if more info are needed, happy to support.

gianlucasullazzo, 2024-06-27:
Hi, same here. It works for some minutes after reload and then values went to 0 soon. No logs (it seems)

olizafiro, 2024-06-28:
Hi,

All good today so far.
Maybe just isolated issue.

ian-morgan99, 2024-06-28:
I have seen this happens to me in the UK, every time there is a bright sunny day. 
I believe this is a constraint on the SolarEdge server with lots of users checking simultaneously, but I have no means to validate this theory.
Where might it be possible to see any logs to explore the theory ?

Obelix74Git, 2024-06-30:
Same problem here. Yesterday was A very sunny day and I received all day the values. Today is a normal day but didn’t receive anything yet also not in the early morning.
The SolarEdge portal itselfs and HA SolarEdge plug-in are werking fine.
If I look to the logs it juist doesn’t see any value.

Mariusthvdb, 2024-10-28:
any updates here?

only noticed this the other day, but all my panels show 0 for their values

[image]

except the life time energy.

checking the Solar Edge app show correct values there. There have been several updates for the app, but would they have changed the webpage this integration is scraping?

confirm no errors in the logs

rinyheuvelman, 2025-03-30:
Issue still not resolved, restarting HASS helps only until there are no more readings again

Gwr1234, 2025-04-11:
Hi. I have only been using this app for a few days but I have seen this shutting down reported here. All 3 days have been sunny with the values being reported until early evening then everything goes to zero. Restarting the app everything works fine for a short while and then stops again.
Screen shot shows reporting dropping to zero and then resuming after reloading app.
[image]

ian-morgan99, 2025-04-11:
Reloading the integration fixes it for me without a full reboot .
I remain convinced it's a problem at the server end as it seems to always be the sunniest days when people are probably querying the server more , but I have no way to prove it .

Gwr1234, 2025-04-11:
Definitely associated with early evening. Started dropping to zero again at almost same time as yesterday. Reloading app restores it for a short while but fails again.

[image]