ProudElm/solaredgeoptimizers #78: Error on Setup, Jason parsing error?

dbaarda, 2024-10-13:
I'm just trying to set this up. Installed via HACs which says I've installed v1.2.4, but the service self-reports as v1.2.2. After entering siteid, username, and passwd it doesn't seem to find any devices or entities. Looking in the logs I find;

```Python
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:361
integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 9:34:27 AM (2 occurrences)
Last logged: 9:46:45 AM

Error while setting up solaredgeoptimizers platform for sensor
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity_platform.py", line 361, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/home/homeassistant/.homeassistant/custom_components/solaredgeoptimizers/sensor.py", line 97, in async_setup_entry
    info = await hass.async_add_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 64, in requestSystemData
    raise errortje
  File "/srv/homeassistant/lib/python3.12/site-packages/solaredgeoptimizers/solaredgeoptimizers.py", line 56, in requestSystemData
    if json_object["lastMeasurementDate"] == "":
       ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
TypeError: list indices must be integers or slices, not str
```

Is this some kind of typo error in a new release? Or a change to the jason returned?

ProudElm, 2025-03-20:
This is the same issue as #81

closed 2025-03-20 by ProudElm

dbaarda, 2025-04-29:
Just a little followup. I finally got around to fixing this by doing "ADD ENTRY" on the SolarEdge Optimizers Data service, re-entered all the details, and this time it worked, so I deleted the old entry.

I have no idea what the difference was, but various things have been updated on my system since I last tried, and as per #81 maybe there was a spurious space somewhere in my earlier attempt?