ProudElm/solaredgeoptimizers #87: Detected code that calls `device_registry.async_get_or_create` referencing a non existing `via_device` (

Berg0987, 2025-02-19:
Logger: homeassistant.helpers.frame
Source: helpers/frame.py:234
First occurred: February 17, 2025 at 12:39:02 PM (6 occurrences)
Last logged: 3:27:28 PM

Detected code that calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('solaredgeoptimizers', '01JEKS6DZH3D5FW37D9HF1K8SA'), with device info: {'hw_version': '180B036F-95', 'identifiers': {('solaredgeoptimizers', '180B036F-95')}, 'manufacturer': 'Silfab Solar Inc.', 'model': 'SIL-430 QD', 'name': '1.0.1', 'via_device': ('solaredgeoptimizers', '01JEKS6DZH3D5FW37D9HF1K8SA')}. This will stop working in Home Assistant 2025.12.0, please report this issue
Detected code that calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('solaredgeoptimizers', '01JMFY8Y8MXBBBA1MMCZ1PB8YG'), with device info: {'hw_version': '180B036F-95', 'identifiers': {('solaredgeoptimizers', '180B036F-95')}, 'manufacturer': 'Silfab Solar Inc.', 'model': 'SIL-430 QD', 'name': '1.0.1', 'via_device': ('solaredgeoptimizers', '01JMFY8Y8MXBBBA1MMCZ1PB8YG')}. This will stop working in Home Assistant 2025.12.0, please report this issue

avdbij, 2025-03-18:
Detected that custom integration 'solaredgeoptimizers' calls device_registry.async_get_or_create referencing a non existing via_device ('solaredgeoptimizers', '01JNCAKDM3AHTY596BD8E6H351'), with device info: {'hw_version': '204D9050-4D', 'identifiers': {('solaredgeoptimizers', '204D9050-4D')}, 'manufacturer': 'Trina Solar', 'model': '280 Full Black', 'name': '1.0.1', 'via_device': ('solaredgeoptimizers', '01JNCAKDM3AHTY596BD8E6H351')} at custom_components/solaredgeoptimizers/sensor.py, line 103: async_add_entities(. This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/ProudElm/solaredgeoptimizers/issues
2025-03-18 16:27:47.050 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solaredgeoptimizers is taking over 10 seconds.
2025-03-18 16:27:49.487 WARNING (MainThread) [homeassistant.helpers.frame] Detected code that calls device_registry.async_get_or_create referencing a non existing via_device ('solaredgeoptimizers', '01JNCAKDM3AHTY596BD8E6H351'), with device info: {'hw_version': '204D9050-4D', 'identifiers': {('solaredgeoptimizers', '204D9050-4D')}, 'manufacturer': 'Trina Solar', 'model': '280 Full Black', 'name': '1.0.1', 'via_device': ('solaredgeoptimizers', '01JNCAKDM3AHTY596BD8E6H351')}. This will stop working in Home Assistant 2025.12.0, please report this issue

ProudElm, 2025-03-19:
Fixed in merge #86