ProudElm/solaredgeoptimizers #88: Currents of one panel keeps 'unknown'

mariobrosch, 2025-03-04:
All of my other 20 panels work perfectly, but this one always stays on unknown.

[image]

ProudElm, 2025-03-19:
I see the same now.. 

Took a while for me to get back into this, now i have a bit more free time. I will check the issues.

ProudElm, 2025-03-20:
I cant get my dev environment installed because of the bug of #81. First i need to fix that one

vk5ztv, 2025-05-30:
I'm seeing the same here, and it's been there for some time. Happens after a HASS restart, but eventually clears. Looks like the first polled data is missed but everything else is parsed OK - perhaps just a timing issue or race condition when the integration is first started?  Any progress?