ProudElm/solaredgeoptimizers #93: References non-existent device — warning in Home Assistant, will break in 2025.12

Migaddo, 2025-05-04:
I'm seeing the following warning in Home Assistant when using the solaredgeoptimizers custom integration:

Detected code that calls device_registry.async_get_or_create referencing a non existing via_device ('solaredgeoptimizers', '760e67126f726cb9c7e00f308e8e9c69'), with device info: {'hw_version': '10285584-11', 'identifiers': {('solaredgeoptimizers', '10285584-11')}, 'manufacturer': 'Trina Solar Energy', 'model': 'TSM-270DC05A.08', 'name': '1.1.1', 'via_device': ('solaredgeoptimizers', '760e67126f726cb9c7e00f308e8e9c69')}.

This is triggered at:
custom_components/solaredgeoptimizers/sensor.py, line 103

According to the warning, this will become a breaking error in Home Assistant 2025.12 if via_device references are not validated against existing registered devices.

The via_device reference should only be used after the parent device is created in the device registry.

Home Assistant version: 2025.5.x

Integration installed via: HACS

rinyheuvelman, 2025-07-14:
Current home assistant version 2025.7.1.
Hopefully the code will be updated soon within the next five months. 😉😎

rinyheuvelman, 2025-08-22:
Still getting the warning message…