ProudElm/solaredgeoptimizers #94: PSA: https://github.com/home-assistant/core/pull/146036 adds similar functionality in core

tronikos, 2025-06-03:
For a few months now this integration is practically useless for me and it seems for many people judging by all the recent issues. Data from SolarEdge is often delayed by a couple of hours so this integration zero outs the sensors. This integration is also the slowest I have with all the requests it makes.

So in https://github.com/home-assistant/core/pull/146036 I'm adding an integration in core with similar functionality but uses statistics instead of sensors, and a single request to get the energy produced for all equipment over the last 7 days.

8jg55fvhu7558b, 2025-06-15:
Sounds interesting. Would it be possible to use your modified integration right now, even when it is not merged?

tronikos, 2025-06-18:
@ProudElm How many optimizers/modules do you have? I have 27.

DeFlanko, 2025-06-20:
@tronikos  how's this coming along?

tronikos, 2025-06-20:
It's out of my control. PRs in core often take several months before someone starts reviewing them... We need to be patient.

DeFlanko, 2025-06-20:
I just checked the link you shared , is there anything you do for the Python error in the checker? Or is that someone on the HA team?

Berg0987, 2025-07-16:
Is it possible to download your code now and try it before  it gets approved by HA?

Presjar, 2025-07-20:
> Yes you can download the solaredge_modules directory from my branch, copy it under config/custom_components and add version in the manifest.json. For how to use it see https://deploy-preview-39361--home-assistant-docs.netlify.app/integrations/solaredge_modules/

I completed the steps and can now see solaredge_modules integration. Have logged into my account with the setup steps. However I dont get and items showing under Entities .

tronikos, 2025-07-20:
There are no entities/sensors. Read the documentation.

Berg0987, 2025-07-20:
Sorry, not looking for the entities or sensors.  Looking for a link to the location of your code that I need to copy to my instance of Home Assistant.

Gunnar

8jg55fvhu7558b, 2025-08-11:
The same applies to me – I couldn't find your integration in version 2025.8.0 of HA.
I would very much like to test this new integration. Hopefully it will be officially available soon.

billraff, 2025-08-24:
Will your project make it to September's release? I'd ask in a thread on that project's page but can't find it.

tronikos, 2025-08-26:
Very unlikely to make it to the September release. The cut is in 2 days. So in the meantime I created https://github.com/tronikos/solaredge_modules