ProudElm/solaredgeoptimizers #96: device_registry.async_get_or_create

brembygit, 2025-08-16:
Detected that custom integration 'solaredgeoptimizers' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('solaredgeoptimizers', '<GUID>'), with device info: {'hw_version': '17D25DF3-39', 'identifiers': {('solaredgeoptimizers', 'ID)}, 'manufacturer': 'SunPower', 'model': 'SPR-MAX3-430', 'name': '1.2.1', 'via_device': ('solaredgeoptimizers', '<GUID>')} at custom_components/solaredgeoptimizers/sensor.py, line 103: async_add_entities(. This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/ProudElm/solaredgeoptimizers/issues

Ulrich72, 2025-08-17:
Same here:

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:350
First occurred: 08:23:26 (1 occurrence)
Last logged: 08:23:26

Detected that custom integration '**solaredgeoptimizers**' calls `**device_registry.async_get_or_create**` referencing a non existing `**via_device**` ('solaredgeoptimizers', '8ab31faffc44e3c23ba877eb3c857cfb'), with device info: {'hw_version': '142337B4-22', 'identifiers': {('solaredgeoptimizers', '142337B4-22')}, 'manufacturer': 'Hengdian Group DMEGC Magnetic Co. Ltd.', 'model': 'DM405M10-54HBB-V', 'name': '1.1.1', 'via_device': ('solaredgeoptimizers', '8ab31faffc44e3c23ba877eb3c857cfb')} at **custom_components/solaredgeoptimizers/sensor.py, line 103**: async_add_entities(. 

**This will stop working in Home Assistant 2025.12.0**, please create a bug report at https://github.com/ProudElm/solaredgeoptimizers/issues

rinyheuvelman, 2025-08-30:
Same warning:
`Detected that custom integration 'solaredgeoptimizers' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('solaredgeoptimizers', 'b773395bb22a7209dd33e4c1089e0262'), with device info: {'hw_version': '10F0880F-97', 'identifiers': {('solaredgeoptimizers', '10F0880F-97')}, 'manufacturer': 'Canadian Solar', 'model': '330', 'name': '1.1.1', 'via_device': ('solaredgeoptimizers', 'b773395bb22a7209dd33e4c1089e0262')} at custom_components/solaredgeoptimizers/sensor.py, line 103: async_add_entities(. This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/ProudElm/solaredgeoptimizers/issues `

rinyheuvelman, 2025-09-22:
`Detected that custom integration 'solaredgeoptimizers' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('solaredgeoptimizers', 'b773395bb22a7209dd33e4c1089e0262'), with device info: {'hw_version': '10F0880F-97', 'identifiers': {('solaredgeoptimizers', '10F0880F-97')}, 'manufacturer': 'Canadian Solar', 'model': '330', 'name': '1.1.1', 'via_device': ('solaredgeoptimizers', 'b773395bb22a7209dd33e4c1089e0262')} at custom_components/solaredgeoptimizers/sensor.py, line 103: async_add_entities(. This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/ProudElm/solaredgeoptimizers/issues `